llvm/circt #5: Implement lowerings from various FIRRTL nodes to rtl.concat

lattner, 2020-06-20:
The RTL dialect and verilog printer support an `rtl.concat` operation.  The LowerToRTL.cpp pass should lower `firrtl.shl(x, 4)` ==> `{x, 4'h0}`, `firrtl.cat`, `firrtl.cvt` etc to it.

lattner, 2020-06-20:
@drom is looking into this

drom, 2020-06-22:
About `firrtl.shl(x, 4) ==> {x, 4'h0}` convertion:
VerilogRTL has shift operators:
* logical shift operators, `<<` and `>>`
* arithmetic shift operators, `<<<` and `>>>`
Should we have them in the RTL Dialect?

drom, 2020-06-22:
About `firrtl.cvt`

### FIRRTL
[image]

### Verilog RTL
[image]

Should we have `$signed` operation type in RTL Dialect?

lattner, 2020-06-22:
Yeah, it is a good question about whether we have signed or unsigned types in RTL dialect.  It doesn't really make sense for a hardware oriented dialect to do that, but given the importance of Verilog in practice, it might make sense.

For now, lets stick with signless types.  If it becomes a problem in practice we can always revise this.

lattner, 2020-06-22:
Also xref for general background (not saying this is the right thing to do here necessarily): https://mlir.llvm.org/docs/Rationale/Rationale/#integer-signedness-semantics

stephenneuendorffer, 2020-06-29:
Vivado HLS uses LLVM-style signless types for most of the compiler.   This causes some subtleties in range analysis which tracks a signed range and an unsigned range for each value.  In RTL, signed operations are typically represented like:

  wire   [8:0] add_ln235_fu_662_p2;
  reg   [8:0] j_0_reg_360;
  assign add_ln235_1_fu_687_p2 = ($signed(9'd482) + $signed(j_0_reg_360));

(In other words, most everything is declared as vectors of bits and signed operations are expressed by always converting the operands to signed.  There are a few places where wires are declared as signed, but this is the exception more than the rule.)

lattner, 2020-10-14:
I think this is complete now!

closed 2020-10-14 by lattner