llvm/circt #7: Implement lowering of firrtl.wire to rtl.wire

lattner, 2020-06-22:
We now have an rtl.wire operation, and the verilog printer supports it as well, but we should implement lowering from firrtl.wire.

There are also other lowering that can be done: connect and partial connect, node, etc but those can be handled as a follow-on task.

lattner, 2020-06-22:
Amalee is planning to take this on. I don't have her github ID though.

amaleewilson, 2020-06-22:
Commenting for Github username

amaleewilson, 2020-06-25:
@drom Can you give me write access for this repo?

amaleewilson, 2020-06-25:
Not sure if this belongs here or in another issue, but I'll work on connect next. Also need to implement vectors in the RTL dialect to handle, e.g., !firrtl.vector<uint<1>, 4>.

amaleewilson, 2020-06-29:
When lowering FIRRTL connect to RTL connect, RTL connect wants the source and destination to have the same types, but the FIRRTL dialect often has a flipped type, e.g. !firrtl.flip<uint<4>>. Should this be left as a separate todo for handling flips or does it make sense to just convert the flipped thing into a nonflipped thing, e.g. !firrtl.flip<uint<4>> to !firrtl.uint<4>?

lattner, 2020-06-29:
We should eliminate flip when lowering types.  "input" and "output" and "inout" are all hints in verilog, they don't carry semantic meaning.  We should eventually have a way to express this when we have an rtl.module, but for now it is safe to drop these.

amaleewilson, 2020-07-01:
@fabianschuiki  raised a point about handling bidirectionality, like inout, and suggested considering this more in depth.

fabianschuiki, 2020-07-01:
> @fabianschuiki raised a point about handling bidirectionality, like inout, and suggested considering this more in depth.

It may also be worthwhile to expressly *not* support this, and adopt a single-definition notion for the RTL dialect. This would make optimizations easier I figure, because figuring out what value a wire can assume remains a local problem.

lattner, 2020-07-02:
I'd recommend that we talk about this in a design meeting - the tradeoffs are nuanced.

ekiwi-sifive, 2020-08-06:
> It may also be worthwhile to expressly _not_ support this, and adopt a single-definition notion for the RTL dialect. This would make optimizations easier I figure, because figuring out what value a wire can assume remains a local problem.

In `firrtl` we have the `Analog` type which can only be connected to in order to support "tunneling" unknown wires between ports and Verilog blackboxes.

lattner, 2020-09-12:
Closing this out as the main lowering has been implemented.  More specific issues can be filed for other cases.

closed 2020-09-12 by lattner