llvm/circt #18: Add support for lowering vectors of wires from FIRRTL to RTL

amaleewilson, 2020-06-28:
While FIRRTL seems to have only vectors of wires, Verilog allows n-dimensional wires, which can be either packed or unpacked. The packed vs unpacked distinction has a huge impact on simulator performance. From my discussions with Verilog folks, here are some examples of wires:

```
wire a; // single wire, e.g. firrtl.wire : !firrtl.uint<1>
wire [7:0] b; // bitvector, e.g. firrtl.wire : !firrtl.uint<8> (I think) 
wire [0:0] c; // bitvector length one, not equivalent to a. 
wire [0:0] d [0:7] // unpacked, technically 2D
wire [0:0] [0:7] d // packed, technically 2D
wire [2:0] [0:7] e // packed, 2D
wire [2:0] [0:7] e [3:0] // unpacked?, 3D

q = b[5]; // slice one bit of bitvctor b
r = b[5:5] // vector slice of bitvector b

// more examples from http://www.verilogpro.com/verilog-arrays-plain-simple/
reg [31:0] x      [127:0];        // 128-element array of 32-bit wide reg
wire[15:0] y[7:0], z[7:0];        // 2 arrays of 16-bit wide wires indexed from 7 to 0
reg [ 7:0] mema   [255:0];        // 256-entry memory mema of 8-bit registers
reg        arrayb [  7:0][255:0]; // two-dimensional array of one bit registers
```

I'm not sure how fine grained these distinctions should be in the RTL dialect, but it might make sense to start with vectors and eventually support tensors of wires.

lattner, 2020-06-28:
Interesting, is the 'packed vs unpacked' distinction just simulator performance, or is there something else going on here?

It seems like there are a couple of ways we could model this:
1) we could model this with a list of dimensions, then a "number of packed dimensions" integer in the type.
2) we could model this as a single "packed vector" and "unpacked vector" type that compose onto each other.

I think the major distinction should be how these compose with non-bit types like the SystemVerilog bundle types.

solomatnikov, 2020-06-28:
FYI https://www.veripool.org/issues/1656-Verilator--RFC-Relaxing-UNOPTFLAT-by-dividing-unpacked-array#note-1

Something to keep in mind when generating arrays b/c Verilator is the most used open source simulator.

drom, 2020-06-29:
```verilog
//    2     1        3     4
bit [1:5] [1:6] v4 [1:7] [1:8];
```
4D array
1,2 = packed dimentions
3,4 = unpacked dimentions

1 varies most rapidly, followed by 2,3,4

stephenneuendorffer, 2020-06-29:
Generally for memory inference to work you want something like:
reg [DWIDTH-1:0] ram[0:MEM_SIZE-1];

amaleewilson, 2020-06-29:
@lattner I think option 2 is most intuitive.

drom, 2020-06-29:
All N **packed** dimensions have to compose first before to compose with M **unpacked** dimensions.

lattner, 2020-06-29:
It could be good to bring this up on the forum to see if others have an opinion, it would also be worth looking at what LLHD is doing here.

teqdruid, 2020-10-08:
I'm not sure if we ended up discussing this in one of the meetings, but I'm planning on adding a [tensor type to the RTL dialect](https://llvm.discourse.group/t/optimizations-and-the-rtl-dialect/1923/10?u=jdd). I welcome additional input!

lattner, 2021-01-03:
We now have ArrayType (packed) and UnpackedArrayType, I think this is done.

closed 2021-01-03 by lattner