llvm/circt #31: linker error building from master

drom, 2020-07-08:
I am getting linker errors while building from master

```
/usr/bin/ld: lib/libMLIRHandshakeOps.a(HandshakeOps.cpp.o): in function `ValueRange<llvm::SmallVector<mlir::Value, 4> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
clang-10.0: error: linker command failed with exit code 1 (use -v to see invocation)
```

```
ldl  -ltinfo  -lpthread  -lm  ../llvm/build/lib/libLLVMDemangle.a && :
/usr/bin/ld: lib/libMLIRLLHDToLLVM.a(LLHDToLLVM.cpp.o): in function `ValueRange<std::array<mlir::Value, 2> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
/usr/bin/ld: lib/libMLIRLLHDToLLVM.a(LLHDToLLVM.cpp.o): in function `ValueRange<std::array<mlir::Value, 5> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
/usr/bin/ld: lib/libMLIRLLHDToLLVM.a(LLHDToLLVM.cpp.o): in function `ValueRange<std::array<mlir::Value, 7> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
/usr/bin/ld: lib/libMLIRLLHDToLLVM.a(LLHDToLLVM.cpp.o): in function `ValueRange<std::array<mlir::Value, 1> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
/usr/bin/ld: lib/libMLIRLLHDToLLVM.a(LLHDToLLVM.cpp.o): in function `ValueRange<std::array<mlir::Value, 3> &, void>':
/home/drom/work/github/circt/circt/build/../llvm/mlir/include/mlir/IR/OperationSupport.h:890: undefined reference to `mlir::ValueRange::ValueRange(llvm::ArrayRef<mlir::Value>)'
clang-10.0: error: linker command failed with exit code 1 (use -v to see invocation)
```

Anybody experiencing similar issues?

lattner, 2020-07-08:
I am seeing similar build problem,s I just nuked my build dir, updated submodules and am trying again.

lattner, 2020-07-08:
A clean build worked for me just now, you might need to update your git submodules

drom, 2020-07-08:
@lattner thank you. It worked for me.

closed 2020-07-08 by drom