llvm/circt #33: canonicalization of RTL variadic operations

drom, 2020-07-08:
## clump

## fold
https://mlir.llvm.org/docs/Canonicalization/#canonicalizing-with-fold

* [ ] `and(..., 0) -> 0`
* [ ] `and(..., '1) -> and(...)`
* [ ] `and(..., x, x) -> and(..., x)`

* [ ] `or(..., 0) -> or(...)`
* [ ] `or(..., '1) -> '1`
* [ ] `or(..., x, x) -> or(..., x)`

* [ ] `xor(..., 0) -> xor(...)`
* [ ] `xor(..., '1) -> not(xor(...))`
* [ ] `xor(..., x, x) -> xor(...)`

## identities

uenoku, 2020-10-16:
Is there NotOp in RTL dialect now?(Sorry if I am missing)

lattner, 2020-10-16:
`not(x)` is expressible as `xor(x, -1)`

uenoku, 2020-10-16:
Thanks, since `not` is in the list above, I thought it is defined as Op somewhere.

drom, 2020-12-05:
we should have `rtl.not` Op #300

drom, 2021-03-25:
LGTM

closed 2021-03-25 by drom