llvm/circt #70: Implement Affine-To-Handshake

stephenneuendorffer, 2020-08-17:
Converting Affine loops directly to the handshake dialect gives the opportunity to use affine analysis of array index expressions to determine that access points are independent. This can result in more more memory parallelism.

kumasento, 2020-08-19:
Hi @stephenneuendorffer 

I'm currently working on this, while there is always a question popped up in my mind: should we convert affine to handshake directly, or convert affine to standard first and then reuse the standard-to-handshake pass? 

Intuitively, the first option gives us more flexibility, but there might be much logic overlapped with the existing standard-to-handshake pass, since the body of `affine.for` loop may contain standard ops.

I would prefer the second option because:

1. We can reuse the existing affine-to-standard (in MLIR) and standard-to-handshake conversions.
2. We can add analysis and canonicalization passes from the affine dialect for handshake before the conversion happens (I'm not sure how the analysis results can be passed down to the follow-up passes, but I suspect it won't be that hard). 

If there is no assumption that is obviously wrong, I will go ahead with the second option and try to put together a simple prototype asap. Meanwhile, I'm curious about what the independent test you've mentioned looks like. Would you mind pointing me a simple test case? Thanks!

stephenneuendorffer, 2020-08-19:
I think the big question is how to keep dependence information between memory access operations around after AffineToStandard, so that it can be used by StandardToHandshake.  I think doing this is challenging, so this would be the biggest reason to go straight from the Affine dialect to the Handshake Dialect.  Otherwise just doing a progressive lowering is pretty straightforward using the existing transformations.

As an example, here is a little bit of code with 4 memory accesses:
```
for (i = 1; i < N; i++)
    x[i] = x[0] + x[i]*y[i];
```
All of these memory accesses are, in fact, independent, assuming x and y don't alias.  All of the accessess on the right side of the statement can be shown to access different locations, and the write to x[i] never writes to a location that the read x[i] accesses in the loop.  Taking this information into account allows us to build a better handshake design with more pipelining possible.

kumasento, 2020-08-19:
Thank you very much Steve!

> I think the big question is how to keep dependence information between memory access operations around after AffineToStandard, so that it can be used by StandardToHandshake. I think doing this is [...]

This makes perfect sense. Thank you for pointing this out. I currently have no idea about how to propagate dependence information down to the following passes neither. So maybe we should go with the direct conversion way.

> As an example, here is a little bit of code with 4 memory accesses:
>
> ```
> for (i = 1; i < N; i++)
>     x[i] = x[0] + x[i]*y[i];
> ```
>
> All of these memory accesses are, in fact, independent, assuming x [...]

Thank you for this example! It makes things much clearer. In affine there are utilities to analyse such cases. I will make it as one test case to verify my future implementation.