llvm/circt #80: [firrtl] Implement more verifier checks for the FIRRTL dialect

lattner, 2020-09-12:
There are a bunch of invariants spelled out in the FIRRTL spec that aren't enforced by the MLIR verifier.  For example, it is invalid for the dest of a firrtl.connect to be smaller than the source of it.  These should be enforced by the firrtl dialect so compiler bugs are caught closer to the source.

vasantteja, 2020-09-15:
Can I assign this issue?

stephenneuendorffer, 2020-09-15:
Please do!

seldridge, 2020-09-21:
Note: some of the invariants may be missing from the spec. As an alternative, all the checks that matter should be found in the following files. Implementing checks that work for any of the exceptions contained within would be great first issues.

- [CheckHighForm.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/CheckHighForm.scala)
- [CheckInitializations.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/CheckInitialization.scala)
- [CheckTypes.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/CheckTypes.scala)
- [CheckWidths.scala](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/CheckWidths.scala)

lattner, 2021-06-20:
@seldridge is this done now?

seldridge, 2021-06-21:
The spirit of this is done. Any further tracking should be more narrow verification of SFC/MFC error behavior. E.g., there are a lot of exceptions that the SFC can throw related to a failure in its CheckHighForm that should be verified do the same thing in the MFC and provide equivalent or better error messages.

closed 2021-06-21 by seldridge