llvm/circt #82: [RTL] Implement comparisons

lattner, 2020-09-12:
The FIRRTL dialect has things like `firrtl.lt` but there are no corresponding RTL dialect ops for them.  The ops should be added and lowerings implemented.

Note that the FIRRTL ops encode the sign of the operation into the types of the operands.  We should use MLIR/LLVM style comparisons that embed the signedness into the operation name themselves, e.g. like [std.cmpi](https://mlir.llvm.org/docs/Dialects/Standard/#stdcmpi-cmpiop).

stephenneuendorffer, 2020-09-12:
If it's 'like std.cmpi', is there a reason why we don't just use std.cmpi?   My preference would be to leverage the standard dialect whenever possible for core representations that aren't related to an external language (like FIRRTL).

lattner, 2020-09-13:
I guess we could do that, but it makes me uneasy for several reasons:

1) The standard dialect can go away, several people are unhappy with it existing.
2) The standard dialect is strongly associated with IRs that use imperative control flow, not implicitly parallel constructs like HW.
3) We'd end up with it mixed into RTL constructs, which would then make progressive lowering to imperative dialects messier - because the std operations would have one semantic when "in rtl" and one semantic "after lowering".
4) we're likely to have slightly different operations (perhaps not for comparisons) e.g. the rtl arithmetic operations are variadic, and there is a proposal to allow mixed width operands.  Having things change between std and rtl based on that seems weird to me.

stephenneuendorffer, 2020-09-14:
I find it hard to see how the purely functional semantics of 'add' change between sequential semantics and concurrent semantics.  Certainly FuncOp implies sequential semantics, as does scf.for, but I see little reason for duplicating 'add' in lots of dialect, not to mention duplicating the optimization rules for 'add'.  Allowing variadic arguments I think is generally a good thing, and in particular will simplify some aspects of tree balancing for arithmetic operations,  but as you note, this isn't particularly useful for comparisons.
I'm not opposed to adding 'basic' ops in these dialects, but I think there should be a clear reason why existing ops are not appropriate and we should leverage composability with existing dialects whenever possible.  If those dialects go away, we can cross that bridge when we come to it.

lattner, 2020-09-14:
Do you understand my point about progressive lowering?  This is a really big issue.  If you'd like, we can talk about it in the meeting on wednesday.

An `rtl.cmpi` and an `std.cmpi` also differ in that one is "designed" to work with cyclic graphs etc, the other isn't.  Small differences like this will probably magnify over time.

stephenneuendorffer, 2020-09-14:
I guess I don't understand your point.  To me, rtl.cmpi and std.cmpi are exactly the same.  They take a set of input values and generate a set of output values.  The differences in when they do that, where the values come from, and what happens to the result values are completely determined by the region what they exist in, either an SSACFG region or a graph region, and the semantics of the operation that encloses that region.

I'd appreciate better understanding the design difference that you see that need to be reflected in these operations, but I will probably miss the meeting this week, unfortuantely.

I think there might be value in having both so that conversions must be implemented as a complete lowering, rather than a partial lowering, but I see this as an unclear benefit, given the additional transformation rules that would have to be implemented to do this.

lattner, 2020-09-18:
We didn't discuss this yesterday, but we did discuss one related issue: the treatment of zero width wires.

Zero width wires are very useful for generator frameworks (e.g. the bottom'ing out of recursion, handling the addressing bits for a one-element memory etc).  In an "implicitly extending" IR, you need to define what a zext/sext from zero bits means, or somehow cordon them off somewhere.  FIRRTL has partial support for this sort of thing, but it is apparently adhoc and not fully baked.

darthscsi, 2020-10-08:
Addition: https://reviews.llvm.org/D89059
Lowering coming soon.

lattner, 2020-10-14:
Is this done @darthscsi ?

darthscsi, 2020-10-16:
Yes it is, at least for integers.

closed 2020-10-16 by darthscsi