llvm/circt #83: Question about MLIR FileCheck.

cgyurgyik, 2020-09-13:
Hi all. I am still ramping up to CIRCT/MLIR, and have been stuck on this for a while now. I'll explain what I think is happening, and you can proceed to tell me why I'm wrong :). If there's a better location for questions like this one, please let me know and I'll go there. Thanks!

So I added the following bit of code to the ```AndOp::getCanonicalizationPatterns``` in ```Dialect/RTL/Ops.cpp```

```
      // and(..., x, x) -> and(..., x) -- idempotent
      if (inputs[size - 1] == inputs[size - 2]) {
        rewriter.replaceOpWithNewOp<AndOp>(op, op.getType(),
                                           inputs.drop_back());
        return success();
      }
```

Now, I'd like to actually see that the input is actually dropped within ```test/rtl/bitwise.mlir```.
I wrote the following:

```
// CHECK-LABEL: func @and_equal_arguments_removed(%arg0: i7, %arg1: i7) -> i7 {
func @and_equal_arguments_removed(%a: i7, %b : i7) -> i7 {
// CHECK-NEXT:    [[RES0:%[0-9]+]] = rtl.and %arg0, %arg1, %arg1 : i7
// CHECK-NEXT:    return [[RESULT:.*]] : i7
  %and1 = rtl.and %a, %b, %b : i7
  return %and1 : i7
}
```
^ This passes, but from my understanding, it shouldn't. Instead, the 3rd line should be changed to:
```
// CHECK-NEXT:    [[RES0:%[0-9]+]] = rtl.and %arg0, %arg1 : i7
```
, which says that the last ```%arg1``` was dropped. 
So, there's obviously something fundamentally flawed with my understanding here.

lattner, 2020-09-14:
Hi @cgyurgyik: the simplest explanation is that the pattern isn't matching and the rewrite not happening.  If you set a break point on the `replaceOpWithNewOp` call, is it getting hit?

Make sure the canonicalization patterns are getting registered correctly.

-Chris

cgyurgyik, 2020-09-14:
No, ```replaceOpWithNewOp``` is not getting hit, which is why I thought I was fudging something simple up.
By registered, do you mean it sets the following: [hasCanonicalizer = 1](https://github.com/llvm/circt/blob/85824588348303c894ecffb236294c8ea145f75b/include/circt/Dialect/RTL/Combinatorial.td#L82) ?

lattner, 2020-09-14:
Yes, and you need to add some code to register it in `getCanonicalizationPatterns` for the op: `results.insert<YourFolder>(context);`.

cgyurgyik, 2020-09-14:
I believe this is also already occurring here: https://github.com/llvm/circt/blob/85824588348303c894ecffb236294c8ea145f75b/lib/Dialect/RTL/Ops.cpp#L213

lattner, 2020-09-14:
Ok, then I'm not really sure why it isn't triggering - once it does, the filecheck logic should work fine :-).  I'd step through the code in a debugger to figure out what isn't happening as you expect.  For example, is the `if (inputs[size - 1] == inputs[size - 2]` line getting reached, etc.

cgyurgyik, 2020-09-14:
Ok so I ended up using a constant as the last two arguments, and it works fine now. Thanks for the back-and-forth!

closed 2020-09-14 by cgyurgyik

lattner, 2020-09-14:
Great!