llvm/circt #90: [RTL Dialect] Teach `rtl.wire` to do fancy things with SSA names

lattner, 2020-09-19:
The firrtl dialect does some fancy things with SSA names: it pulls the 'name' attribute into the result of the operation, and correctly round trips this through the parser.  The RTL dialect should do something similar, but limited to `rtl.wire` specifically.  Instead of printing:

```
%42 = rtl.wire { name = "foo" } : i8
```

We should print:

```
%foo = rtl.wire : i8
```

The real payoff of this is in the connects and other things that use the wire as an operand.

The code that does this is the `FIRRTLOpAsmDialectInterface` stuff in lib/Dialect/FIRRTL/Dialect.cpp, it should be straight-forward to adapt and simplify it for RTL dialect.

cgyurgyik, 2020-09-20:
I'm probably missing something here, but isn't this the same thing as what's already being tested here:
https://github.com/llvm/circt/blob/8c3d96d3a8a7e8495c248928ace193a9607510d6/test/rtl/basic.mlir#L40

lattner, 2020-09-20:
The problem with this is that we parse this, but discard the name, so circt-opt doesn't print `%w` if you round trip it.  We should instead recognize the name, store it in the IR as a 'name' attribute on the wire, then print the name attribute back out.

uenoku, 2020-10-13:
Hi, I created PR for this :)

jatincse, 2020-10-28:
I want to contribute in this ,please guide me as i am new in this organization.

lattner, 2020-11-23:
@uenoku fixed this some time ago, thanks!

closed 2020-11-23 by lattner