llvm/circt #107: Lower FIRRTL module to RTL module

amaleewilson, 2020-09-28:
Now that an RTL module exists, we can lower FIRRTL modules to RTL modules.

Since the lower-to-rtl pass runs on a FIRRTL module, would it make sense to change the lower-to-rtl pass to run on a FIRRTL circuit, make RTL modules, lower the FIRRTL stuff into them, and then remove the FIRRTL modules? Or is there some better way of doing this? I'd like to leave as much of the existing pass in place as possible.

lattner, 2020-09-29:
Yeah I'd love to see this.  I'm not sure the best way to handle this.  The existing lowering pass is parallel across `FModuleOp`s, so it isn't supposed to touch the enclosing `CircuitOp`.  I think it would make sense to keep this, but then add a subsequent pass over the `CircuitOp` that lowers the modules.  This allows parallelism across the module level, and then a quick cleanup pass that lowers the modules themselves.

lattner, 2020-09-29:
@River707 does this seem sensible to you?  Any precedent we should be aware of here?

fabianschuiki, 2020-10-10:
I was wondering the same thing for the FIRRTL-to-LLHD pass. It currently runs on `CircuitOp`, but running in parallel on `FModuleOp` and doing a cleanup pass as you suggest is a great idea.

lattner, 2020-10-14:
I'm going to take a look at this

lattner, 2020-11-09:
Finally landed in fcf8f8b !

closed 2020-11-09 by lattner