llvm/circt #123: [FIRRTL] Add support for lowering aggregate types within FIRRTL

mikeurbach, 2020-10-05:
Based on discussion [here](https://llvm.discourse.group/t/firrtl-aggregate-types-discussion/1806), we would like to add some lowering passes to turn high- and mid-FIRRTL constructs into low-FIRRTL constructs. This is one such pass.

There was already some discussion in https://github.com/llvm/circt/issues/18 about lowering aggregates, in the context of lowering to RTL. The pass I'm proposing would do this lowering within FIRRTL, so aggregates will already be flattened before lowering to RTL or LLHD.

From Discourse:

> Flattening of aggregates is a property of low FIRRTL IR currently being handled by [LowerTypes](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/LowerTypes.scala). This is pretty straightforward.

I'm proposing to add a similar pass to CIRCT.

lattner, 2020-10-14:
Yeah agreed, we need to implement all of the lowering passes from CHIRRTL to Low FIRRTL.

mikeurbach, 2020-10-14:
I have a start of this lowering pass that seems to be working for flattening bundle types. I'll clean it up and make a PR. It should be straightforward to extend the approach to work with vectors and subindexes, but I haven't really looked at supporting everything in the Scala implementation.

At this point, I'm able to lower Handshake -> FIRRTL -> low(er)FIRRTL -> RTL -> EmitVerilog. EmitVerilog still doesn't support `firrtl.when`, but this gets us a step closer to having the layers work end-to-end. I'm still checking if it's actually the System Verilog that was intended, but I think it's looking pretty good. Here's my test:

[example.txt](https://github.com/llvm/circt/files/5380963/example.txt)

seldridge, 2020-10-15:
Example output is looking good. Let me know when you have the PR ready to go (or a draft PR) and I can take a look.

If you're looking for some example test cases for this, you may be able to re-use the existing Scala FIRRTL Compiler `LowerTypes` tests in: https://github.com/freechipsproject/firrtl/blob/master/src/test/scala/firrtlTests/LowerTypesSpec.scala

Note: as of recently the Scala FIRRTL Compiler does "uniquification" as part of type lowering. E.g., if you have both `a: {b: UInt<1>}` and `a_b: UInt<1>` in the same module, then it mangles the former, during lowering, to `a__b`. Previously, this "uniquification" was a separate pass and it was assumed that types could be blindly lowered. I assume there's some clever existing way to get a new, non-conflicting name (in a symbol table, maybe?)?

mikeurbach, 2020-10-15:
RE: uniquification, I was taking a look at that in the Scala implementation, and I'll see what we need here. I'd done some tests where I accidentally generated flattened ports with conflicting names, and it seemed like MLIR was noticing that and generating non-conflicting names automatically. I'll make sure to have a test for this explicitly.

commit 15dedc11 (2020-10-21): [FIRRTL] Add start of FIRRTL types lowering

This adds a new pass to lower FIRRTL types, initially focused on
bundle types. This flattens bundle types in module and instance
ports, and deals with the flattened values in subfield and connect
operations. Starts to resolve #123.
commit 73909884 (2020-10-23): [FIRRTL] Add start of FIRRTL types lowering

This adds a new pass to lower FIRRTL types, initially focused on
bundle types. This flattens bundle types in module and instance
ports, and deals with the flattened values in subfield and connect
operations. Starts to resolve #123.
commit 81edcc3b (2020-11-05): [FIRRTL] Add start of FIRRTL types lowering

This adds a new pass to lower FIRRTL types, initially focused on
bundle types. This flattens bundle types in module and instance
ports, and deals with the flattened values in subfield and connect
operations. Starts to resolve #123.
PR #157 (2020-11-16): [FIRRTL] Add start of FIRRTL types lowering
This adds a new pass to lower FIRRTL types, initially focused on
bundle types. This flattens bundle types in module and instance
ports, and deals with the flattened values in subfield and connect
operations. Starts to resolve #123.
closed 2020-11-16 by mikeurbach