llvm/circt #163: [FIRRTL] module instances can appear before module declarations

darthscsi, 2020-10-20:
Output from firrtl can have module instances before module declarations.  firtool rejects these files.

seldridge, 2020-10-20:
Thanks for filing this. I've been bitten by this when trying to produce FIRRTL and then re-parse it with firtool. 

Note: I did open a Scala FIRRTL Compiler issue to add a custom transform that will sort modules so you can get them in a format that firtool won't barf on: https://github.com/freechipsproject/firrtl/pull/1905

I'm not against changing this in the FIRRTL spec as def before use seems sound. This also simplifies some Scala FIRRTL compiler checks of instance cycles (which become impossible since FIRRTL has no forward declarations).

lattner, 2020-10-20:
Ah interesting, I guess the scala implementation does a topo sort before emitting the .fir files but that is not required.

lattner, 2021-06-21:
Fixed in 3d9ce440, huzzah

lattner, 2021-06-21:
It'd be worth taking a look at the FIRRTL project regression tests that dependent on this to see if they shake out any bugs now.

closed 2021-06-21 by lattner