llvm/circt #174: [RTL] Improve the asm syntax of RTLModuleOp

teqdruid, 2020-10-21:
Currently, the syntax is:
```mlir
rtl.module @B(%a: i1 { rtl.inout }) -> (i1 {rtl.name = "b"}, i1 {rtl.name = "c"}) 
```

With the not-really-optional port names for the output ports as hacky attributes on the results. Ideally, we'd have something like what was suggested in #93. Unfortunately (from #169): 

> On RTLModuleOp asm format: Ideally, the syntax would include the port names in the result specifications as shown in the #93. This would (best I can tell) require re-implementing the parseFunctionSignature helper in MLIR to modify the format of the result section. parseFunctionSignature is actually broken into two parts (args and results) internally but sadly those two functions are not exposed publicly. If anyone can think of a less heavyweight option than copy-pasting from MLIR, let me know!

lattner, 2020-10-22:
Yeah, great minds think alike, it would be very nice to improve here.

lattner, 2020-11-23:
Implemented in d8f7e5f

closed 2020-11-23 by lattner