llvm/circt #178: [Handshake] Update `HandshakeToFIRRTL` to use low-level logic instead of `WhenOp`

mikeurbach, 2020-10-23:
For more information, see original Discourse discussion: https://llvm.discourse.group/t/handshake-lowerings-for-merge-like-and-branch-ops/2054.

These ops are all similar in that they deal with merging and branching, and their implementation is simplified by using the FIRRTL `WhenOp`. For the reasons discussed, it makes sense to implement these operators directly using low-level logic instead of `WhenOp`.

I plan to update the lowerings for each of the following ops (in separate PRs):
- [ ] MuxOp
- [ ] MergeOp
- [ ] ControlMergeOp
- [ ] ConditionalBranchOp

mikeurbach, 2020-10-25:
@hanchenye @stephenneuendorffer I started taking a look at this, and I have a question about the ops that implement the `MergeLikeOpInterface`. In these ops, it seems like we should apply the `SameOperandsAndResultType` trait. We don't have that right now, but does that constraint make sense here?

hanchenye, 2020-10-25:
@mikeurbach I think the select signal of `MuxOp` has different type with other operands.

mikeurbach, 2020-10-25:
Ah, true. I guess I am specifically referring to the data operands (as returned by the `dataOperands` method on `MergeLikeOpInterface`). These operands should always have the same type (and have the same type as the result), correct?

hanchenye, 2020-10-25:
Yes, I think that's correct!

mikeurbach, 2020-11-13:
@hanchenye I have been looking into `MergeOp` and `ControlMergeOp` next, and I just want to check in before digging too far into the implementation. I reviewed Section 3.4 and Figure 7 [here](http://www.cs.columbia.edu/~sedwards/papers/edwards2019compositional.pdf). My understanding is Handshake's `ControlMergeOp` op is what they implement in Figure 7., i.e. it reports the index of the selected input as an output.

I had previously mentioned I would start with implementing `MergeOp` since it is simpler, but now I'm thinking it would be advantageous to start with the trickier `ControlMergeOp`. Then we can factor out the common part, and just leave off the unneeded aspects when implementing `MergeOp`. If that all sounds good I'll plan on sharing a PR for `ControlMergeOp` next.

mikeurbach, 2020-12-14:
@hanchenye I think we can close this now.

closed 2020-12-14 by mikeurbach