llvm/circt #183: [FIRRTL] Parser fails on negative integer parameters

seldridge, 2020-10-26:
Signed integer parameters will cause the FIRRTL dialect parser to fail. 

Consider the following FIRRTL snippet:

```
circuit Foo:
  extmodule Bar:
    parameter A = -1
  module Foo:
    inst bar of Bar
```

When run through the Scala FIRRTL compiler, you get the following Verilog:

```verilog
module Foo(
);
  Bar #(.A(-1)) bar (
  );
endmodule
```

When run through `firtool` you get:

```bash
firtool Foo.fir
# Foo.fir:3:19: error: expected '.' in floating point literal
#    parameter A = -1
#                  ^
```

This also may motivate a change in the underlying data format of integer parameters. Clearly, something like `-1` needs to be set as a signed integer. However, should unsigned parameters continue to be stored as signless types, should they be unsigned, or should they be signed?

lattner, 2020-10-27:
If using an `APInt` (like MLIR attributes are) the typical convention is to store them as signless types, sign/zero extended to their native width.  In this case it should be all ones in either case.

LLVM also has an `APSInt` type that tracks sign internally, so you can just use a divide operator and it knows which one to do.  I don't think that MLIR attributes expose that though.

lattner, 2020-12-25:
Fixed in 362a282, thanks @seldridge !

closed 2020-12-25 by lattner