llvm/circt #201: `ninja install` doesn't install binaries such as `circt-translate`

GeorgeLyon, 2020-11-04:
Installing only creates `include` and `lib` directories. It should also create `bin`.

I poked around and tried to figure out what was different between CIRCT's CMakeLists.txt, and that of LLVM and MLIR, but I couldn't figure out which magic incantation we were missing. Maybe someone with a more seasoned understanding of how the build is set up can point me in the right direction.

stephenneuendorffer, 2020-11-04:
I think the issue is that the binaries are defined as LLVM 'tools', but CIRCT does not include defaults for all of LLVM's CMake configuration options.  Normally tools are built by default.  I attempted to refactor this here: https://reviews.llvm.org/rGe9b87f43bde8b5f0d8a79c5884fdce639b12e0ca but I haven't had a chance to come back to this patch.

GeorgeLyon, 2020-11-04:
Hmm, I assumed the reason for this was that eventually we may want to bring the CIRCT dialects "in-tree" so using the LLVM constructs would be considered a plus (I've also wondered why circt wasn't just an additive fork of LLVM, this could mitigate some of the weirdness around having LLVM as a submodule). If that isn't the case then yes, we should probably stop co-opting the LLVM operations.

closed 2020-11-06 by GeorgeLyon