llvm/circt #206: [FIRRTL] LowerToRTL Unhandled op failure

darthscsi, 2020-11-05:
Consider:
module {
  firrtl.circuit "foo" {
    firrtl.module @bar(%io_cpu_flush: !firrtl.uint<1>) {}
    firrtl.module @foo() {
      %fetch = firrtl.instance @bar {name = "fetch"} : !firrtl.bundle<io_cpu_flush: flip<uint<1>>>
      %1454 = firrtl.subfield %fetch("io_cpu_flush") : (!firrtl.bundle<io_cpu_flush: flip<uint<1>>>) -> !firrtl.flip<uint<1>>
      %hits_1_7 = firrtl.node %1454 {name = "hits_1_7"} : !firrtl.flip<uint<1>>
      %1455 = firrtl.asPassive %hits_1_7 : (!firrtl.flip<uint<1>>) -> !firrtl.uint<1>
    }
  }
}

FIRRTLLowering::handleUnloweredOp inserts a StdIntCast to a flipped type which fails to verify.

mikeurbach, 2020-11-06:
I think I was running into this while testing https://github.com/llvm/circt/pull/157. I wasn't sure what to do about it, so I'm glad to see you've opened this issue. Let me know if i can help out.

darthscsi, 2020-11-06:
@mikeurbach, you can see if the pr helps you out.

PR #207 (2020-11-06): Can't cast to a non-passive type, need another node
Closes #206
closed 2020-11-06 by darthscsi