llvm/circt #208: Improve the Circt build system

GeorgeLyon, 2020-11-06:
I wanted to start a discussion about how we can improve the circt build system, please let me know if there is a better venue than this issue.

During the (admittedly brief) time I've been working with Circt, I've experienced pain and seen others experience pain related to the build system. Some concrete examples include: the Xcode generator not working, a two-step build process, binary artifacts not being installed. My gut feeling for why this is the case is that Circt lives with one foot in and one foot out of the LLVM build system. @stephenneuendorffer opined [here](https://github.com/llvm/circt/issues/201) that we could improve this by moving further away from LLVM, but I feel like LLVM has reasonable solutions for a bunch of stuff in the CMake world and moving away risks us reinventing the wheel and making a potential future reconciliation with LLVM/MLIR more difficult. 

My alternative suggestion is that we buy into the LLVM system _more_, not less. In this world, Circt is would just be a fork of LLVM with a new "circt" sub project which in the file system is a sibling to `llvm` and `mlir`. The main drawback to this approach is that we no longer have a clear separation between LLVM commits (which currently end up in the `circt/llvm` fork) and `circt` commits, but I think we can effectively mitigate this by creating a GitHub action that only passes for commits that are either (1) entirely additive in directories we have "blessed" as being part of circt (2) do not touch _any_ circt files or folders, and thus can eventually be upstreamed to LLVM verbatim. 

The benefits of this approach are that (hopefully) having llvm drive the build will give us all the goodness that llvm provides without having too roll our own solutions or debug partially-adopted llvm solutions. The build also becomes single-stage at this point so a simple 
```
$ cd circt
$ mkdir llvm/build
$ cd llvm/build
$ cmake -G Ninja ../llvm \
    -DLLVM_ENABLE_PROJECTS="circt" \
    -DLLVM_TARGETS_TO_BUILD="X86;RISCV" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG
$ ninja
$ ninja check-circt
```
will do the right thing. One other consequence of having a single build step is that you could work with MLIR and Circt in a single workspace in IDEs that support cmake integration (like VSCode).

There are also no submodules to manage with this approach, updating the submodule would be replaced by simply merging new commits from the upstream LLVM. Since the circt commits are required to only be additive there should be no LLVM-circt merge conflicts. This will also get rid of the weirdness of having `llvm/circt` depend on `circt/llvm` (I'd support the `circt/llvm` fork simply becoming the new repository, though we can also manually make `llvm/circt` a fork of `llvm`), and make [this](https://github.com/llvm/circt/pull/188) a non-issue.

All that said, I understand that this may be disruptive to `circt`'s community of contributors and we should weight the impact of that versus the benefits of simplifying the build.

lattner, 2020-11-06:
@compnerd Do you have any thoughts about how to improve this, or do you know another expert-level llvm/cmake person who would make sense to talk to?

stephenneuendorffer, 2020-11-06:
Hi George,

In most of these respects, CIRCT attempts to follow best practices with MLIR and out-of-tree projects that depend on LLVM.
There are some disadvantages of the current approach, but there are also some clear advantages, at least including:
1) CMAKE configuration does not require configuring LLVM.
2) precommit tests run fast, because LLVM is cached.
3) Clear separation between LLVM/MLIR modifications and CIRCT modifications.

In particular, we really want to avoid the sense that we are 'forking' LLVM.  This is definitely *NOT* the goal.  If circt/llvm is becoming a problem, then we can remove it and depend on upstream LLVM.  The main downside of this is that it may slow down CIRCT development as the small number of MLIR changes we need are upstreamed first, or make developers less willing to push changes to MLIR, so we end up with redundancies in CIRCT.

To be clear, the direction that I think CMake integration needs to move is not by separating from LLVM, but by building LLVM CMake mechanisms that embrace the fact that there are lots of dependent projects of LLVM that live in-tree and out-of-tree. Most projects already do this to some extent, because supporting any form of non-trivial build (you want a shared library, and then a extension that links into python?) needs to take many aspects of how LLVM is linked into account.

I agree it would be nice to have CMake integration in VScode working more conveniently.  I'm relatively new to VScode and I've never been able to get it working, even for a standalone LLVM build.

stephenneuendorffer, 2020-11-06:
@GeorgeLyon To solve your original problem with the tools not being installed, I think something like this in circt/CMakeLists.txt is a short-term fix.

option(LLVM_INCLUDE_TOOLS "Generate build targets for the LLVM tools." ON)
option(LLVM_BUILD_TOOLS "Build the LLVM tools. If OFF, just generate build targets." ON)

teqdruid, 2020-11-06:
I hear your pain, but I agree with @stephenneuendorffer 100%. I've found the circt fork of LLVM to be helpful as I can stage my MLIR patches there and use them in CIRCT. As we are a new project targeting a fundamentally different computational device, I expect there to be a non-trivial number of those patches.

I'll also note that the LLVM/MLIR folks are actively looking at modern alternatives to their build/review/CI/etc systems. They're looking for experiences from their subprojects. By being a parallel subproject with a submodule pointing to it, we contribute to that experience in the form of participating in those discussions and pushing relevant changes upstream.

That said, there are a number of things we should improve -- #188 being an important one. We couldn't get consensus on that one since there's no one person in charge of cmake/repo/CI/etc. (Well, I suppose I'm responsible for the build & test workflows.) I think we could solve this issue by having one person "own" the developer stuff. I don't think any of us are cmake experts, which is where I personally run into most of the issues.

stephenneuendorffer, 2020-11-06:
I'm happy to push on making progress with build flow issues.  I'm far from a CMake expert, but I've dug into the MLIR CMake quite a bit, so I mostly understand how it's intended to work. :)

compnerd, 2020-11-06:
@lattner - from my experience with Swift, I would say that if there are dependencies on LLVM, buying heavily into the LLVM build system is valuable.  This becomes even more valuable if you are trying to create additional tools that you would want to ship into a single toolchain.  If there is a build time dependency on more than say LLVMSupport, the integration can actually reduce overall build times (due to parallelism exposed and less duplication of platform checks).  The simplicity and parallelism isn't entirely free - it does restrict your ability to mix and match build types of LLVM (that is, you can not build LLVM in release and CIRCT in debug).  If that is valuable enough to justify this or not is a judgement call and one that the project would need to make.

One question that is likely to come up - do you really need CMake (and from the thread here, seems to already have).  I would say that depends on what you are trying to do.  If you fully control the environment and do not care about any deviation from that, then another option would be GN which lets you define the build without the need for CMake, but makes integration with other projects and of other projects with this project more difficult.  CMake has made some decisions which can be frustrating to work with, but it is still better than most other build systems for flexibility of target environments and integration and most projects have adopted it as the standard build system.

One thing to note is that doing the conversion into the LLVM system is probably cheaper right now than later.  As LLVM only recently migrated to the monorepo layout, doing an incremental roll-over is possible where you can keep CIRCT out of the LLVM tree and use the `LLVM_EXTERNAL_PROJECTS`, `LLVM_<PROJECT>_SOURCE_DIR` options to "inject" the project into the LLVM build and support using a standalone LLVM build as a staging mechanism.

For the concern of build times - the build can limit to exactly what you need to build rather than building all of LLVM if you are using a unified build.  Because the project is built as a single entity, the build system is able to see a complete graph and only build the targets that are used.

I would say that @llvm-beanz probably is a good person to speak to as well as he did most of the LLVM build migration and is generally pretty knowledgeable about CMake.

stephenneuendorffer, 2020-11-06:
@compnerd To me the time that CMake takes to run is just as significant as the overall build time, because cmake does not incrementally generate the build system and is not parallelized.   CIRCT uses cmake information exported from LLVM through LLVMConfig.cmake and leverages the same core CMake infrastructure as LLVM.  This is the process generally followed by projects dependent on LLVM + MLIR.

compnerd, 2020-11-06:
@stephenneuendorffer Using LLVMConfig is absolutely one way to do this - its making a different set of tradeoffs.  It works well for having a static LLVM that is prebuilt and rarely changes.  By doing that, you can avoid rebuilding LLVM.  The approach that I suggested works well for cases where LLVM and CIRCT would be more intrinsically tied (and would potentially reduce the CMake time).

llvm-beanz, 2020-11-06:
I think @compnerd has hit pretty much all the major points I can think of, except maybe a thought on longer term project goals. If CIRCT’s goal is to be included as an LLVM project in the main repository, full integration with the LLVM build system should be a requirement of that. That doesn’t mean you can’t support a standalone build using LLVMConfig (as Clang and LLDB do), but it does mean that the general proposal to buy _more_ into the LLVM CMake build system is the right decision.

@stephenneuendorffer, I also am not sure I agree with your perspective that CMake time is as significant. Clean CMake runs are significantly slower than regenerations, and while it is true that CMake doesn’t run truely incrementally or in parallel, a full regeneration is rarely needed. I have profiled LLVM’s CMake in the past and fixed low hanging fruit that caused slow regeneration, and if that is a significant problem we can work to address it. Fixing performance issues in LLVM’s CMake is a much better long term solution than reducing reliance on LLVM’s build system, especially in the monorepo where there is a significant push to have more tightly coupled tooling.

teqdruid, 2020-11-06:
> It works well for having a static LLVM that is prebuilt and rarely changes

This is exactly how CIRCT currently is.

> If CIRCT’s goal is to be included as an LLVM project in the main repository, full integration with the LLVM build system should be a requirement of that.

I don't think that's necessarily our goal though there may be differing opinions. It hasn't been discussed much. In general, I support out-of-tree builds but there are advantages to being in the monorepo. I don't think all of the dialects in CIRCT and supporting software are appropriate for the monorepo, though I don't know what the standard is. I could imagine *eventually* lifting and shifting the core CIRCT dialects into the monorepo if they're not too intertwined with the other CIRCT dialects. (This would ensure that changes to MLIR don't break our paradigms.) @stephenneuendorffer @lattner Would y'all like to weigh in on this?

stephenneuendorffer, 2020-11-06:
I think with the current build system, CIRCT could drop-in to the monorepo next to other sub-projects with minimal modifications.  I would like to see CIRCT grow into an end-to-end compilation flow for hardware, just as clang and flang exist for software flows.  e.g. adding something like:

# If we are not building as a part of LLVM, build Clang as an
# standalone project, using LLVM as an external library:
if( CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR )

Actually, there is probably nothing preventing us from doing this today and supporting LLVM_EXTERNAL_PROJECTS="circt"?

stellaraccident, 2020-11-06:
A couple of random comments from skimming the thread:

* I think that if the trajectory is towards the monorepo, setting things up so that the default dev flow involves driving from the LLVM cmake via LLVM_EXTERNAL_PROJECTS (or nesting CMake files to a similar effect) is a good step. There are benefits to the install split as is done but there are also drawbacks -- it is mainly a mode of operation that you don't want to lose through neglect, etc (since it is important for downstream packaging tasks that will come into play later). While IREE's approach is somewhat more involved, it does support either and defaults to a build that is fully integrates LLVM and IREE's code into one CMake/ninja build. This has the advantage that you can touch any file in the main (IREE) repo or the LLVM (submodule) and ninja will do the right thing, which is pretty convenient for the average development flow. Then you mainly need to think about how to stage things between the two when setting things up for review and submit, which if you have multiple repos will always involve some thought and recipes. I'm going to end up changing NPCOMP soon to not build with the install split by default because for some platforms, we need more customization to what LLVM builds vs what can be managed with a static LLVM install, and living in the same CMake invocation is pretty much the only way to do that. I still want to retain the ability of separate installs, though, as an option.

* Some people love VSCode's CMake integration. I personally hate it and find it incredibly intrusive, even when I take the exaggerated steps to get it configured right. When moving across projects, I prefer to do the minimal setup and just drive CMake and ninja from the command line. You can then take one of two options that get you to a functioning environment:

  1. Disable CMake auto-configure and point it at your build file. Then if you configure from the command line, you can at least open the various CMake panels in VSCode and interact with targets, etc.
  2. Just configure Intellisense directly to point at the generated `build_commands.json` in each build directory.

I personally do option #2 almost exclusively and take the time for option #1 if I'm going to be living with a workspace for a while. The first thing I do when adding a folder is open the Intellisense settings and setting the path to `build_commands.json` and then everything (I want) works. There are occasional hiccups and a `CTRL-P > Reload Window` clears them up. I've documented this setup [here](https://github.com/llvm/mlir-npcomp#configure-vscode).

My two cents in case if helpful...

lattner, 2020-11-07:
> If CIRCT’s goal is to be included as an LLVM project in the main repository, full integration with the LLVM build system should be a requirement of that.

Yes, I think this is a long term goal - we should look to incubator projects having a path to "graduate".  If they don't, they shouldn't be incubator projects.

GeorgeLyon, 2020-11-09:
I've added a new issue that discusses LLVM_EXTERNAL_PROJECTS in isolation from some of the other ideas presented here (like repo reconfiguration)

GeorgeLyon, 2020-11-10:
I'm closing this issue, as I now thing there work to support LLVM_EXTERNAL_PROJECTS is a meaningful step forward. We can revisit further improvements after folks have played around with LLVM_EXTERNAL_PROJECTS.

closed 2020-11-10 by GeorgeLyon