llvm/circt #213: [ESI] Cosimulation DPI plugin

teqdruid, 2020-11-07:
# Summary

Elastic Silicon Interfaces provides a feature called cosimulation. Cosim in general allows communication between the simulation and software. In the ESI case, it is typed and can be used to build an application and language specific API which is nearly identical to how the real hardware would interface. This allows users to simulate against the actual target software (or some simplification of it), enabling easier co-design.

ESI cosim uses [Cap'nProto](https://capnproto.org/) as a message format and RPC client/server. Capnp was chosen due to its relatively low-overhead encoding/decoding (as compared to Protocol Buffers and ilk) is a good fit for hardware. Using a standard messaging protocol allows users to work with a variety of languages: the Cap'nProto website lists C++, C#, Erlang, Go, Haskell, JavaScript, Ocaml, Python, and Rust as languages which support messages and RPC!

# Merge plan

This feature addition is huge and touches many knowledge domains (RTL, C++, Python, and lit testing at a minimum). And it's nearly done. I've created a draft PR with the WHOLE thing just to provide the overall picture / preview. I'll be dividing this up into several smaller PRs which are more domain specific and easier to review. They may not make sense in isolation but they'll build towards the cosimulation feature.

# Pull requests

I'm planning the following PRs. List is subject to modifications.

- [ ] Cosim README
- [ ] Capn'Proto generation code
- [ ] DPI scaffolding
- [ ] RTL
- [ ] Cosim DPI plugin and RPC server
- [ ] Integration test

# TODOs:

- [ ] Remove hardcoded port

# Notes

- This addition is entirely separate from the MLIR part of CIRCT. Subsequent PRs will add MLIR modeling of cosim and then RTL generation.
- The integration test is (IMO) really cool since I adapted it into llvm-lit. (This may just be a form of Stockholm Syndrome though.)
- This code is adapted and greatly improved from a prototype (which is based on code @darthscsi wrote while at MSFT). I may have missed a good deal of stylistic issues.

stephenneuendorffer, 2020-11-07:
Looking forward to this 👍

teqdruid, 2020-11-07:
@stephenneuendorffer Do you think this is the right way to stage this particular feature?

stephenneuendorffer, 2020-11-07:
I think it's probably a good place for a PR with a series of patches in it, that way people can review parts, or the whole thing as they see fit.

lattner, 2020-11-07:
Thanks John, I really appreciate you staging in the pieces individually.  It is best for me to understand the parts in pieces.

teqdruid, 2020-11-21:
This finally landed! Thanks everyone for the code reviews.

PR #259 (2020-11-21): [ESI] Cosimulation integration tests
The last PR in the series for #213. Closes #213.

This is the integration test -- a simple loopback check.
closed 2020-11-21 by teqdruid