llvm/circt #223: [FIRRTL] Support for comments?

lattner, 2020-11-09:
The scala implementation of FIRRTL already has partial support for comments (See DescriptionAnnotation and "Declaration descriptions" in the Verilog output tests).  Should we support this in the FIRRTL dialect, as an optional attribute on the associated ops?

seldridge, 2020-11-15:
For a more verbose description of what this looks like, you can see the following Scastie snippet:
  - https://scastie.scala-lang.org/seldridge/l3vDB27xSza1MuuHtqZUWA/20

This is using the two types of `DescriptionAnnotation`, `DocStringAnnotation` and `AttributeAnnotation`, and tries to show how these are passed to the Scala FIRRTL Compiler. This is probably a reasonable test case to prototype how annotations could be converted to something in the FIRRTL Dialect like parameters or attributes. This probably means that the FIRRTL Dialect should be able to injest both a FIRRTL IR file and an Annotation JSON file.

The one thing to watch out for here is that any annotation may be associated with a specific _instance_ of a module and not with all modules. Part of the conversion from FIRRTL IR + Annotations to FIRRTL Dialect would then require generation of parameters (or duplication of the same module) or attributes. This needs more thought/prototyping.

lattner, 2020-11-19:
Agreed, this will be interesting.  I'd recommend starting with a "first principles design" for this sort of thing at the RTL level, then figure out how much we can map onto that from FIRRTL.  I suspect FIRRTL will need something bigger and crazier to handle the general case (if we care) in fullness of time.