llvm/circt #225: Enable a single-stage build using LLVM_EXTERNAL_PROJECTS setting

GeorgeLyon, 2020-11-09:
As one outcome of the discussion here: https://github.com/llvm/circt/issues/208, it seems making it possible to build circt using the LLVM_EXTERNAL_PROJECTS would be a great step forward. It seems to me that this can be done in a way that does not break the the existing two-step build[citation needed]. I created a new issue so we can discuss this in isolation from other ideas brought up in the original issue.

Here is some thin documentation I found for this setting: https://reviews.llvm.org/D20838, and a similarly thin description exists here: https://llvm.org/docs/CMake.html

The following naive command **does not** do the right thing, instead complaining that some MLIR targets are not yet defined:
```
mkdir build
cd build
cmake -G Ninja ../llvm/llvm \
    -DLLVM_ENABLE_PROJECTS="mlir" \
    -DLLVM_TARGETS_TO_BUILD="X86;RISCV" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DLLVM_EXTERNAL_PROJECTS=Circt \
    -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR=..
```

Can someone point me in the right direction to get this working?

stephenneuendorffer, 2020-11-09:
Did you make the appropriate changes to the toplevel circt/CMakeLists.txt to support this?

GeorgeLyon, 2020-11-09:
No, I changed nothing. If someone could point me to some documentation about what that should look like (or at least an example of a project using this setting), I would appreciate it (I'm not a cmake expert by any stretch of the imagination)

stephenneuendorffer, 2020-11-09:
take a look at clang/CMakeLists.txt:

```
cmake_minimum_required(VERSION 3.13.4)

if(POLICY CMP0075)
  cmake_policy(SET CMP0075 NEW)
endif()

# If we are not building as a part of LLVM, build Clang as an
# standalone project, using LLVM as an external library:
if( CMAKE_SOURCE_DIR STREQUAL CMAKE_CURRENT_SOURCE_DIR )
  project(Clang)

  #  Whole bunch of code

  set( CLANG_BUILT_STANDALONE 1 )
  set(BACKEND_PACKAGE_STRING "LLVM ${LLVM_PACKAGE_VERSION}")
else()
  set(BACKEND_PACKAGE_STRING "${PACKAGE_STRING}")
endif()

# A whole other bunch of code

```

Basically, the if condition checks that the project is being built 'standalone' (i.e. the way circt is currently built).  In this case, this CMakeLists.txt is the toplevel cmake file and there's a whole bunch of extra setup code that has to happen.  If not, then this CMakeLists.txt is being included through the LLVM_EXTERNAL_PROJECTS configuration and the extra setup code is not necessary.  So, basically, the current circt/CMakeLists.txt needs to be refactored to have this check.  I *think* that's all that is necessary.

GeorgeLyon, 2020-11-09:
^ I started a PR to track my progress

closed 2020-11-11 by GeorgeLyon