llvm/circt #230: `-DBUILD_SHARED_LIBS` option breaks the build

GeorgeLyon, 2020-11-11:
I tried this at @teqdruid's suggestion here: https://github.com/llvm/circt/pull/227#discussion_r520866983

I ended up with a broken build with the following output: 
[output.txt](https://github.com/llvm/circt/files/5520937/output.txt)

It looks like there might be a missing dependency somewhere, but I'm not familiar enough with LLVM to track it down.

teqdruid, 2020-11-11:
Yeah, I ran into that as well IIRC so I only run it on the LLVM config.

stephenneuendorffer, 2020-11-11:
I'll try to take a look at this.  Can't guarantee I'll get to it tomorrow, though.

GeorgeLyon, 2020-11-11:
It seems like `-DLLVM_BUILD_LLVM_DYLIB=ON` should accomplish a similar goal, at least from the point of view of Circt. @stephenneuendorffer feel free to close this issue unless you still want to investigate.

stephenneuendorffer, 2020-11-11:
There are a number of combinations here.  I theory they should all work.  Basically: LLVM can be built as static libraries or as dynamic libraries.  The LLVM libraries may or may not be built into libLLVM.so, which may or may not be linked into MLIR.  MLIR itself has the same options and may or may not be build into libMLIR.so, etc.  The last time I looked at this in depth, cmake got confused when linking against libMLIR.so, and currently when we build packages against this, we build with static libraries and libLLVM.so, but not libMLIR.so.

PR #241 (2020-11-12): Updates for BUILD_SHARED_LIBS=on
Closes #230 

Generally speaking, building with BUILD_SHARED_LIBS requires more careful description of the dependencies between libraries in cmake, because shared libraries cannot have unresolved symbols.  Generally speaking, this is just a matter of adding the libraries where missing symbols live.
closed 2020-11-12 by stephenneuendorffer