llvm/circt #232: [FIRRTL] Invariants on connect/partialconnect src/dest are not checked

lattner, 2020-11-11:
These are some of the most common ops in practice, it would be great to get some verification checks in here so we can catch compiler bugs earlier in the pipeline.

mikeurbach, 2020-11-18:
I'd love to see this as well. I've found myself mentally verifying these invariants when working with passes that generate FIRRTL. I'm a little nervous there are already places that generate invalid connects. Is anyone taking a look at this?

If no one else is working on this already, I'd be interested in contributing here. I'm reviewing sections 4.5, 4.6, 5.1, 5.2, and 8 in the FIRRTL [spec](https://github.com/freechipsproject/firrtl/blob/master/spec/spec.pdf). Are those the relevant invariants, and is this the source of truth for the spec?

seldridge, 2020-11-18:
Yes, that's the relevant section. 

If you want some code to cross reference this with, the Scala FIRRTL Compiler is checking connect validity [here](https://github.com/freechipsproject/firrtl/blob/cd845bdbfea0c09e9edbf61e651ede5197d8a084/src/main/scala/firrtl/passes/CheckTypes.scala#L152). Interval types are something that's not in the spec and don't have any wide use (you can ignore those). The real check for connect is then in [this method](https://github.com/freechipsproject/firrtl/blob/ecb96e83324ea17cf38b7b90753d745d3c7f51bd/src/main/scala/firrtl/WIR.scala#L260). 

For partial connect, the code reference is [here](https://github.com/freechipsproject/firrtl/blob/cd845bdbfea0c09e9edbf61e651ede5197d8a084/src/main/scala/firrtl/passes/CheckTypes.scala#L123).

Note that Chisel3 will, by default, never emit a partial connect and will expand all Chisel3 bulk connects into individual connects. Therefore, verifying connect is probably higher priority. 

This is a great thing to tackle! Thanks for offering to take a look. I expect there's a ton of low hanging fruit here, e.g., connecting a `UInt` to an `SInt` currently isn't caught :grimacing::

```
# cat Foo.fir
circuit Foo:
  module Foo:
    input a: UInt<1>
    output b: SInt<1>

    b <= a
```

Scala FIRRTL Compiler output:
```
# firrtl -i Foo.fir
Exception in thread "main" firrtl.passes.CheckTypes$InvalidConnect: : [module Foo]  Type mismatch in 'b <= a'.
  b: SInt<1>
  a: UInt<1>
```

`firtool` output:
```
# firtool Foo.fir  
module {
  firrtl.circuit "Foo" {
    firrtl.module @Foo(%a: !firrtl.uint<1>, %b: !firrtl.flip<sint<1>>) {
      firrtl.connect %b, %a : !firrtl.flip<sint<1>>, !firrtl.uint<1>
    }
  }
}% 
```

### Getting a Scala FIRRTL Compiler

I find it useful to compare whatever `firtool` is doing with what the Scala FIRRTL Compiler does. The best way to get the latter is by using [Coursier](https://get-coursier.io/). With that, you can then do the following to give you a binary, `firrtl-1.4.0` that you can run like above:

```bash
cs bootstrap edu.berkeley.cs::firrtl:1.4.0 -M firrtl.stage.FirrtlMain -o firrtl
```

mikeurbach, 2020-11-18:
Awesome, thanks for the pointers. I will definitely get a working `firrtl` to compare against.

Agreed that focusing on `connect` first should be priority. Besides what Chisel3 is doing, it seems like the Handshake lowering within CIRCT is only using `connect` as well.

Incidentally, I just caught a bug in my own pass producing FIRRTL connects that violated conditions 3. and 4. in section 5.1 of the spec. I'll certainly be adding verification for those conditions :)

mikeurbach, 2020-11-18:
@seldridge I've begun looking into this, and first things first, I'm porting over the comparison routine for checking type equivalence.

One thing is throwing me off with ResetTypes. This [comment](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/WIR.scala#L258-L259) says

> ResetType can be driven by itself, Bool, or AsyncResetType, but it cannot drive Bool nor AsyncResetType. 

That part seems clear, but the [implementation](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/WIR.scala#L266-L267) is what is confusing me:

```
      case (ResetType, tpe)                 => legalResetType(tpe)
      case (tpe, ResetType)                 => legalResetType(tpe)
```

The first line seems to match what the comment said, but doesn't the second line mean ResetType can drive Bool or AsyncResetType? I'm familiar with Scala but by no means an expert, so perhaps I'm simply misunderstanding the logic.

CC @jackkoenig who appears to have done the original implementation.

jackkoenig, 2020-11-18:
Sorry, the comment is wrong. `ResetType` can drive concrete reset types `UInt<1>` and `AsyncResetType`.

This appears to be an example of a comment that got out of sync with the code (despite being part of the same PR). The PR where I implemented reset inference* was a long lived branch (April - Aug 2019) so things apparently got out of sync. Mea culpa.

*inference implies a lot more sophistication than is really there, but it's actually pretty simple type inference.

mikeurbach, 2020-11-18:
Got it, thanks for the quick reply! I will implement the same logic in CIRCT.

lattner, 2020-11-19:
awesome, thank you for tackling this mike!

mikeurbach, 2020-11-23:
I have a question about the orientation of fields in bundle types. Section 4.5 of the spec states:

> the bundles’ i’th fields have matching names and orientations

The Scala implementation appears to be checking that [here](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/WIR.scala#L281) when it says `f1.flip == f2.flip`.

My understanding is this means both fields must be flipped, or not flipped. Is that correct?

The reason I ask is in many places within CIRCT where we lower from Handshake to FIRRTL, we have exactly the opposite logic. For example, [here](https://github.com/llvm/circt/blob/master/test/Conversion/HandshakeToFIRRTL/simple_addi.mlir#L44). Should this and all the other connections between bundles that Handshake generates be illegal? Am I missing something?

mikeurbach, 2020-11-23:
I've been digging into this with the Scala `firrtl` implementation, and I have a feeling that what we are generating from Handshake here may be wrong.

For example, here is some FIRRTL:

```
circuit Test:                                                                                                                                                                                                      
  module Test:                                                                                                                                                                                                     
    input in: {valid: UInt<1>, flip ready: UInt<1>, data: UInt<8>}                                                                                                                                                 
    output out: {flip valid: UInt<1>, ready: UInt<1>, flip data: UInt<8>}                                                                                                                                          
    out <= in                                                                                                                                                                                                      
```

If I try to run that through `firrtl`, I get:

```
Exception in thread "main" firrtl.passes.CheckTypes$InvalidConnect: : [module Test]  Type mismatch in 'out <= in'.                                                                                                 
  out: { flip valid : UInt<1>, ready : UInt<1>, flip data : UInt<8>}                                                                                                                                               
  in: { valid : UInt<1>, flip ready : UInt<1>, data : UInt<8>}                                                                                                                                                     
```

If instead I update that to the following:

```
circuit Test:                                                                                                                                                                                                      
  module Test:                                                                                                                                                                                                     
    input in: {valid: UInt<1>, flip ready: UInt<1>, data: UInt<8>}                                                                                                                                                 
    output out: {valid: UInt<1>, flip ready: UInt<1>, data: UInt<8>}                                                                                                                                               
    out <= in                                                                                                                                                                                                      
```

`firrtl` doesn't complain.

It seems to me there has been a misunderstanding in the Handshake lowering, and it was assumed fields should have opposite orientations when bundles are connected together. My understanding of the FIRRTL spec is the fields must have the same orientations, because individual fields are flipped or not flip with respect to the overall flow of the bundle. In the valid example above, `in` is a source and `out` is a sink, so flipping the `ready` field in both effectively makes `in.ready` a sink, and `out.ready` a source, when the bundles are connected. The details of how bundles are lowered to ground types takes care of flipping the connect around (Section 5.1.1):

> Connect statements between two bundle typed components connects the
i’th field of the right-hand side expression and the i’th field of the left-hand
side expression. If the i’th field is not flipped, then the right-hand side field
is connected to the left-hand side field. Conversely, if the i’th field is flipped,
then the left-hand side field is connected to the right-hand side field.

This is already implemented in CIRCT's LowerTypes here: https://github.com/llvm/circt/blob/master/lib/Dialect/FIRRTL/LowerTypes.cpp#L321-L324.

If this is making sense, I will update HandshakeToFIRRTL to generate bundles that will pass the new ConnectOp verification.

mikeurbach, 2020-11-23:
Just pushed a fix to my logic in #278 that exposes the issue.

seldridge, 2020-11-24:
Your read of this is right as far as I can tell. Types are going to be insufficient for _fully_ verifying connectivity without the notion of flow. This assumes that we're trying to stay close to the Scala FIRRTL Compiler...

E.g., if you swap the left-hand-side and right-hand-side in your example:

```
circuit Test:                                                                                                                                                                                                      
  module Test:                                                                                                                                                                                                     
    input in: {valid: UInt<1>, flip ready: UInt<1>, data: UInt<8>}                                                                                                                                                 
    output out: {valid: UInt<1>, flip ready: UInt<1>, data: UInt<8>}                                                                                                                                               
    in <= out
```

This isn't a `CheckTypes` error, but a `CheckFlows` error:

```
# firrtl -i Foo.fir
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckFlows$WrongFlow: : [module Test]  Expression in is used as a SinkFlow but can only be used as a SourceFlow.
firrtl.passes.CheckFlows$WrongFlow: : [module Test]  Expression out is used as a SourceFlow but can only be used as a SinkFlow.
firrtl.passes.PassException: 2 errors detected!
```

I think it's fine to initially, narrowly worry about just checking that the types match (and fixing this in Handshake...). Then it's a follow-on concern about adding flow-style verification.

For more information the split of flip and flow stems from perspective of whether a port is either the port of the current module or of an instance. The types of the ports are the same, but their validity for connection changes.

mikeurbach, 2020-11-24:
Thanks for the sanity check @seldridge.

> I think it's fine to initially, narrowly worry about just checking that the types match (and fixing this in Handshake...)

This is my plan for now...

I've been thinking about this more, in the context of how we are doing things in CIRCT. Now I'm wondering if this is just an example of something that has strayed from the Scala FIRRTL Compiler.

In CIRCT, we don't have the notion of `input` or `output` ports. A module's port is an input by default, and if it is an output, the type gets flipped. For example, this manifests in the [FIRParser](https://github.com/llvm/circt/blob/master/lib/FIRParser/FIRParser.cpp#L2136-L2138) and its [tests](https://github.com/llvm/circt/blob/master/test/FIRParser/basic.fir#L5-L8).

I tried to write the example above in MLIR.

The IR below is clearly wrong... both ports are inputs:

```mlir
firrtl.circuit "test" {
  firrtl.module @test(
    %in: !firrtl.bundle<valid: uint<1>, ready: flip<uint<1>>, data: uint<8>>,
    %out: !firrtl.bundle<valid: uint<1>, ready: flip<uint<1>>, data: uint<8>>
  ) {
  }
}
```

No problem, I will just flip the whole bundle type for `%out` and it will become an output (this is my understanding of what should be done based on the above, so please correct me if I'm wrong!):

```mlir
firrtl.circuit "test" {
  firrtl.module @test(
    %in: !firrtl.bundle<valid: uint<1>, ready: flip<uint<1>>, data: uint<8>>,
    %out: !firrtl.flip<bundle<valid: uint<1>, ready: flip<uint<1>>, data: uint<8>>>
  ) {
  }
}
```

Except, as I learned last week, the CIRCT FIRRTL compiler will [canonicalize](https://github.com/llvm/circt/blob/master/lib/Dialect/FIRRTL/Types.cpp#L420-L423) the representation of `FlipType`s, to allow consistent type comparisons by pointer equality (types are always uniqued in MLIR and can be efficiently compared by pointer). By running the example above through `circt-opt` without any flags, the following IR is produced:

```mlir
module {
  firrtl.circuit "test" {
    firrtl.module @test(
      %in: !firrtl.bundle<valid: uint<1>, ready: flip<uint<1>>, data: uint<8>>,
      %out: !firrtl.bundle<valid: flip<uint<1>>, ready: uint<1>, data: flip<uint<8>>>
    ) {
    }
  }
}
```

And now we have the same form I've been staring at, wondering why the heck my [verification](https://github.com/llvm/circt/pull/278/files#diff-6be8f16b249664b12c96f16587df5fbae2167725d4df3b2d04e075610fcbc9c9R722) failed, and if Handshake has been generating invalid IR this whole time.

Now that I've thought through this example a little more thoroughly, I'm actually less concerned, and I don't think Handshake is doing anything wrong. My conclusion, based on the above, is the logic for checking if `BundleType`s are equivalent in terms of Section 4.5 of the FIRRTL spec should just consider their passive types. After that, I can add logic that takes care of checking flow, and worry about the flips then. For what it's worth, this appears to be what the `InstanceOp` [verification](https://github.com/llvm/circt/blob/master/lib/Dialect/FIRRTL/Ops.cpp#L537) is doing, when it makes sure the instance's result `BundleType` matches the module's.

If this all makes sense, I think all I need to do is remove [this check](https://github.com/llvm/circt/pull/278/files#diff-6be8f16b249664b12c96f16587df5fbae2167725d4df3b2d04e075610fcbc9c9R722-R723).

seldridge, 2020-11-24:
In retrospect, I actually think CIRCT is doing the right thing here. By removing `input` and `output` then element-wise matching of ports in the current module should have opposite flippedness. 

An instance port is the opposite semantic.

This then means that handshake is also right.

I think it is safe to deviate from the FIRRTL Spec and how `CheckTypes` and `WrappedType` implement this. I'm trying to run this by other FIRRTL devs and make sure I'm not missing something.

mikeurbach, 2020-11-24:
Cool. I am taking a look at this right now, and I'm going to push an update to #278 that simplifies the code to verify bundle connections using `getPassiveType`.

mikeurbach, 2020-11-24:
Alright, I've updated all of the type equivalence checking to be based solely on passive types here: https://github.com/llvm/circt/pull/278/commits/d346557bc84b3cd70cfc9b58d50c3395f41bbbcb. This simplified some of the other checks in addition to the checks for bundle fields. The tests are passing again, and I think all is well.

I'll wrap up that PR, then immediately follow it up with PRs to address the remaining TODOs about connect verification.

mikeurbach, 2020-11-24:
Thanks @seldridge for being a sounding board while I work through this.

youngar, 2021-05-21:
With the removal of flip canonicalization in  https://github.com/llvm/circt/pull/944, the connect-op verification was locked down tight. I don't think there is anything remaining to do here.

closed 2021-05-21 by youngar

lattner, 2021-05-21:
nice

mikeurbach, 2021-05-21:
awesome