llvm/circt #235: [FIRRTL] Align dhsl lowering with Scala FIRRTL Compiler

seldridge, 2020-11-11:
The Scala FIRRTL Compiler (SFC) does some width padding of the `dshl` operator which is not done by FIRRTL Dialect lowering, i.e., using `firtool`. Specifically, the padding for a shift is:

1. Sign extend the input up to the max width of the shift
2. Do the shift
3. Sign extend the input up to the width of the output

The current lowering from FIRRTL dialect will, from my quick look, cause an issue when lowering `firrtl.sint` shifted into a larger value because the sign extension won't be applied.

Consider the following circuit:

```
circuit Foo:
  module Foo:
    input a_uint: UInt<2>
    output b_uint: UInt<8>
    input a_sint: SInt<2>
    output b_sint: SInt<8>
    input c: UInt<1>
    b_uint <= dshl(a_uint, c)
    b_sint <= dshl(a_sint, c)
```

The SFC will produce, after the `firrtl.passes.PadWidths` transform (`firrtl -i Foo.fir -cll 'firrtl.passes.PadWidths$:trace' && cat Foo.v`):

```
circuit Foo :
  module Foo :
    input a_uint : UInt<2>
    output b_uint : UInt<8>
    input a_sint : SInt<2>
    output b_sint : SInt<8>
    input c : UInt<1>

    b_uint <= pad(dshlw(pad(a_uint, 3), c), 8)
    b_sint <= pad(dshlw(pad(a_sint, 3), c), 8)
```

And the following Verilog:

```verilog
module Foo(
  input  [1:0] a_uint,
  output [7:0] b_uint,
  input  [1:0] a_sint,
  output [7:0] b_sint,
  input        c
);
  wire [2:0] _GEN_0 = {{1'd0}, a_uint};
  wire [2:0] _GEN_1 = _GEN_0 << c;
  wire [2:0] _GEN_2 = {{1{a_sint[1]}},a_sint};
  wire [2:0] _GEN_3 = $signed(_GEN_2) << c;
  assign b_uint = {{5'd0}, _GEN_1};
  assign b_sint = {{5{_GEN_3[2]}},_GEN_3};
endmodule
```

Using `firtool` we get (`firtool -verilog Foo.fir`):

```verilog
module Foo(
  input  [1:0] a_uint,
  output [7:0] b_uint,
  input  [1:0] a_sint,
  output [7:0] b_sint,
  input        c);

  assign b_uint = a_uint << c;	// Foo.fir:8:{12,15}
  assign b_sint = a_sint << c;	// Foo.fir:9:{12,15}
endmodule
```

lattner, 2020-11-12:
Great catch.  You could fix this in the EmitVerilog pass for the firrtl dialect, but I'm hoping it will fall away in the next few weeks.  Does the LowerToRTL pass do the right thing?

lattner, 2020-11-12:
"firtool -verilog -lower-to-rtl foo.fir" should show it

seldridge, 2020-11-12:
Great thought. To my surprise, this actually works. The code gen leaves something to be desired, but this appears functionally equivalent. Namely, the `firrtl.uint` could get away with not padding `c` and the `firrtl.sint` example would look better with a temporary. That said, these changes are almost nitty.

```verilog
module Foo(
  input  [1:0] a_uint, a_sint,
  input        c,
  output [7:0] b_uint, b_sint);

  wire [7:0] _T;	// Foo.fir:2:3
  wire [7:0] _T_0;	// Foo.fir:2:3

  assign _T = {{5'd0}, {{1'd0}, a_uint} << {{2'd0}, c}};	// Foo.fir:8:{12,15}
  assign _T_0 = {{5{({{1{a_sint[1]}}, a_sint} << {3{c}})[2]}}, {{1{a_sint[1]}}, a_sint} << {3{c}}};	// Foo.fir:9:{12,15}
  assign b_uint = _T;	// Foo.fir:2:3
  assign b_sint = _T_0;	// Foo.fir:2:3
endmodule
```

darthscsi, 2020-11-12:
dsh* are (hopefully) lowered to rtl as sign-preserving width extensions of the arguments to the width of the result.

seldridge, 2020-11-12:
> dsh* are (hopefully) lowered to rtl as sign-preserving width extensions of the arguments to the width of the result.

I thought that's exactly what was happening here. It looks like the only mistake is the sign extension of `c` in the `dhsl` for the `firrtl.sint` case. That's getting an `rtl.sext` which is causing a `3{c}` for the shift amount (which is wrong).

The `$signed` is also left out, but I don't think that actually matters for this case.

lattner, 2020-11-13:
Good catch, I filed the zero extend shift amount issue in f443dda.

I haven't put any work at all into improving the quality of the generated Verilog, I'm trying to get LowerToRTL to be correct and complete.  Once that is the case, the next thing will be to improve the verilog quality.  That can be done in a couple of ways: local improvements in EmitVerilog itself, or through the introduction of more ops in the SV dialect that directly model the weird Verilog semantics.  We may end up with a hybrid approach, but some experimentation will be required.

seldridge, 2020-11-13:
New output looks great. Closing this. Thanks for quickly tackling this, @lattner.

```verilog
module Foo(
  input  [1:0] a_uint, a_sint,
  input        c,
  output [7:0] b_uint, b_sint);

  wire [7:0] _T;	// Foo.fir:2:3
  wire [7:0] _T_0;	// Foo.fir:2:3

  assign _T = {{5'd0}, {{1'd0}, a_uint} << {{2'd0}, c}};	// Foo.fir:8:{12,15}
  assign _T_0 = {{5{({{1{a_sint[1]}}, a_sint} << {{2'd0}, c})[2]}}, {{1{a_sint[1]}}, a_sint} << {{2'd0}, c}};	// Foo.fir:9:{12,15}
  assign b_uint = _T;	// Foo.fir:2:3
  assign b_sint = _T_0;	// Foo.fir:2:3
endmodule
```

closed 2020-11-13 by seldridge