llvm/circt #237: Building with LLVM_EXTERNAL_PROJECTS places headers binaries in `tools`

GeorgeLyon, 2020-11-11:
When I set up the build with `LLVM_EXTERNAL_PROJECTS`, as @compnerd suggested, I was able to make the tests pass, but the artifacts are now in `tools/circt/{tools,include}` as opposed to the top-level `bin` and `include` directories. Is this expected? Is there a way I can force the artifacts to end up in the top-level directory?

stephenneuendorffer, 2020-11-13:
What artifacts are you particularly concerned about?  I see binaries in /bin when I do such a build...

GeorgeLyon, 2020-11-13:
I don't see any headers, the "include" directory only has one subdirectory (llvm), whereas the standalone circt build has a top-level "circt" folder (I did notice the binaries in "bin", so maybe that was a different issue). I brought this up because when migrating a project to use the unified build, I needed to change which directories were included. This isn't a huge deal (the workaround is pretty simple) but I was curious about why this was going on and if it was possible to "promote" the circt headers to the top-level include directory to make dependent project configuration easier. 

I should also note that when installing, both build types do the right thing so again,  this isn't that big of an issue. I'll close this issue because it doesn't seem to be something obvious.

closed 2020-11-13 by GeorgeLyon

stephenneuendorffer, 2020-11-13:
Got it.  I *think* this is intended, but one of the things that is missing is that circt doesn't export its cmake configuration information.  (I.e. you're trying to configure a project that depends on circt, and llvm, and mlir, if I understand correctly?)  Do I understand correctly?

GeorgeLyon, 2020-11-13:
I think we are on the same page

PR #242 (2020-11-18): Add cmake export information so that out-of-tree projects can properl…
…y call "find_package(CIRCT)"

Generally speaking, this should allow

cake -DCIRCT_DIR="<something>/lib/cmake/circt" myprojectdirectory, where <something> is
either a build or install area.  Then in your project cmake file you can have:
find_package(CIRCT REQUIRED CONFIG)

Closes #237