llvm/circt #243: [FIRRTL] Inconsistent int type width

uenoku, 2020-11-13:
In `test/firrtl/test.mlir`, there is the following firrtl dialect IR:

```
firrtl.circuit "Top" {
  firrtl.module @Top(%out: !firrtl.flip<uint>,
                     %b: !firrtl.uint<32>,
                     %c: !firrtl.analog<13>,
                     %d: !firrtl.uint<16>) {
    %3 = firrtl.add %b, %d : (!firrtl.uint<32>, !firrtl.uint<16>) -> !firrtl.uint<32>
    ....
  }
}
```
The resulting type of `%3` is given as `uint<32>` but when we look at `getAddSubResult` it should be `uint<33>` because of a parity bit. 
There are several places this kind of inconsistent use in tests (basically related to add/sub op). Is it allowed or should we just prohibit this in the verifier?

lattner, 2020-11-19:
Good catch, I think the verifier should reject this.  We should incorporate getResultType into the verifier, maybe through a new trait that all FIRRTL primitive get?

lattner, 2021-01-03:
Verifier enhanced in 9899b30, and the testsuite was fixed in previous patches.

closed 2021-01-03 by lattner