llvm/circt #247: [FIRRTL] LowerTypes pass does unnecessary work for arguments that are unchanged

mikeurbach, 2020-11-16:
In the original [PR](https://github.com/llvm/circt/pull/157) to add this pass, lowered arguments get added to the end of the argument list in order to avoid trampling on existing arguments' attributes. One implication of this is that _all_ original arguments (even if they don't change) must also be copied to the end of the list so they can keep their relative order. There is more discussion on this comment in the original PR: https://github.com/llvm/circt/pull/157#discussion_r524443743.

We can optimize this pass so that it doesn't do this unnecessary copying and rewriting for unchanged arguments. It will require more bookkeeping in order to keep things straight, but it should be possible.

lattner, 2020-11-17:
Cool, thank you for investigating this.  I don't think the win is worth a lot of increased complexity.  I'm ok with forgoing this if you don't think there is an elegant way to handle this.

mikeurbach, 2020-11-17:
:+1: I will circle back to this after some other tasks and see if there is a nice clean way to address this. I'm pretty sure it can be done with a little thought and possibly another datastructure for bookkeeping.

lattner, 2021-03-14:
Is this done?

mikeurbach, 2021-03-14:
@lattner not yet. When I was doing the performance tuning on this pass, I thought about trying to change this behavior, but ultimately left it off. The fact that we create a new argument for every original argument makes it simple to preserve the relative ordering of arguments as they get flattened, and it wasn't the performance bottleneck at the time.

lattner, 2021-06-20:
@prithayan or @darthscsi is this done?

prithayan, 2021-06-21:
Yes, Andrew has taken care of this in the new version of LowerTypes.
LowerTypes should insert the new lowered argument to the next appropriate position in the list (instead of appending). 
This makes it possible to lower the argument one level at a type and avoids all the unnecessary lowering.
 https://github.com/llvm/circt/blob/dev/darthscsi/LowerTypesV2/lib/Dialect/FIRRTL/Transforms/LowerBundleVectorTypes.cpp#L405
We will close this issue once the PR is merged.

lattner, 2021-06-21:
great thx!

mikeurbach, 2021-07-29:
Was looking for a different FIRRTL issue, and came across this. After @darthscsi's restructuring, we can close this.

closed 2021-07-29 by mikeurbach