llvm/circt #248: Library Rename

stephenneuendorffer, 2020-11-17:
Most of the libraries are named "libMLIRfoo" and not "libCIRCTfoo".  This is probably bad, since it potentially results in collisions with real MLIR libraries.  At the very least we should consistently use one or the other.

xgupta, 2020-12-24:
I submit a [PR](https://github.com/llvm/circt/pull/357) to close this issue but I am seeing another inconsistency in naming.

In the MLIR-NPComp repository where all dialect's implementations are name as fooDialect.cpp and ops as fooOps. In CIRCT ESI, Handshake and LLHD dialect named them as fooDialect.cpp & fooOps.cpp while FIRRTL, RTL, StaticLogic, and SV named them as Dialect.cpp and Ops.cpp.  So it is good practice to rename them also for consistency because I think It also removes their history from the master branch on GitHub webview.

stephenneuendorffer, 2020-12-25:
Git should be able to track the file renames, so I don't think there is much concern there.  Personally, I prefer having foo/fooDialect.cpp rather than foo/Dialect.cpp, even though there is some redundancy in the file naming, because it makes it less likely that files can be confused for eachother.  I'm actually curious how others feel about this?

lattner, 2020-12-26:
I generally agree, it helps with IDE "jump to file" functionality.

xgupta, 2020-12-27:
in another case, in cmake/modules/AddCIRCT.cmake's  [add_dependencies](https://github.com/llvm/circt/blob/6752760bb93835f3e46b019c7dfb2007fd7c17a8/cmake/modules/AddCIRCT.cmake#L3) function, circt-heardes generated by tablegen is currently name as MLIR${dialect}IncGen. Should we change it to CIRCT${dialect}IncGen?

xgupta, 2020-12-29:
Please ignore the last comment, MLIR-NMcomp is also using MLIR${dialect}IncGen naming convention. So should we close library naming issue resolved in #357?

lattner, 2020-12-29:
I support closing this, any further thoughts @stephenneuendorffer ?

stephenneuendorffer, 2020-12-29:
Yes!  Thanks for doing this!

lattner, 2020-12-29:
great job @xgupta !

closed 2020-12-29 by lattner