llvm/circt #255: [FIRRTL] Crash in lower-firrtl-to-rtl-module after lowering from Handshake

mikeurbach, 2020-11-18:
I have this simple program in simple_addi.mlir that I've been trying to compile through Handshake and FIRRTL:

```mlir
func @add(%a: i8, %b: i8) -> i8 {                                                                                                                                                                                  
  %0 = addi %a, %b: i8                                                                                                                                                                                             
  return %0 : i8                                                                                                                                                                                                   
}
```

I am running this like so:

```sh
cat simple_addi.mlir | bin/circt-opt -pass-pipeline='create-dataflow,canonicalize,lower-handshake-to-firrtl' | bin/firtool -format=mlir -lower-to-rtl -pass-pipeline-crash-reproducer=crash.mlir -pass-pipeline-local-reproducer=true
```

It crashes with the following backtrace:

```sh
firtool: ../llvm/build/install/include/mlir/IR/Types.h:312: U mlir::Type::cast() const [U = circt::firrtl::IntType]: Assertion `isa<U>()' failed.                                                                  
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.                                                                                                                              
Stack dump:                                                                                                                                                                                                        
0.      Program arguments: bin/firtool -format=mlir -lower-to-rtl                                                                                                                                                  
 #0 0x00007fd9aae02b23 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/mikeurbach/circt/llvm/llvm/lib/Support/Unix/Signals.inc:563:13                                                                    
 #1 0x00007fd9aae00cb0 llvm::sys::RunSignalHandlers() /home/mikeurbach/circt/llvm/llvm/lib/Support/Signals.cpp:72:18                                                                                               
 #2 0x00007fd9aae02fe5 SignalHandler(int) /home/mikeurbach/circt/llvm/llvm/lib/Support/Unix/Signals.inc:0:3                                                                                                        
 #3 0x00007fd9ad6b88a0 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x128a0)                                                                                                                                
 #4 0x00007fd9a86acf47 raise /build/glibc-2ORdQG/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c:51:0                                                                                                         
 #5 0x00007fd9a86ae8b1 abort /build/glibc-2ORdQG/glibc-2.27/stdlib/abort.c:81:0                                                                                                                                    
 #6 0x00007fd9a869e42a __assert_fail_base /build/glibc-2ORdQG/glibc-2.27/assert/assert.c:89:0                                                                                                                      
 #7 0x00007fd9a869e4a2 (/lib/x86_64-linux-gnu/libc.so.6+0x304a2)                                                                                                                                                   
 #8 0x00007fd9ada7e3fd circt::firrtl::IntType mlir::Type::cast<circt::firrtl::IntType>() const /home/mikeurbach/circt/build/../llvm/build/install/include/mlir/IR/Types.h:0:3                                      
 #9 0x00007fd9ad4efeba extendOrTruncateFIRRTL(mlir::Value, circt::firrtl::IntType, circt::ImplicitLocOpBuilder&) /home/mikeurbach/circt/build/../lib/Dialect/FIRRTL/LowerToRTL.cpp:77:30                           
#10 0x00007fd9ad4efcd1 tryToFindOutputValue(mlir::Value) /home/mikeurbach/circt/build/../lib/Dialect/FIRRTL/LowerToRTL.cpp:288:9                                                                                   
[...]
```

I've also attached the reproducer here: 
[crash.txt](https://github.com/llvm/circt/files/5561663/crash.txt)

It looks like the function that crashes was added recently, and the solution should be simple. I wanted to file this issue anyway before I started working on it.

lattner, 2020-11-19:
This looks like a LowerToRTL regression that I introduced, I'm happy to take a look at this.

lattner, 2020-11-19:
Ah right, your fix is the right fix here: the verifier should catch the incorrect 'connect', so I think the LowerToRTL logic is correct, thanks!

PR #256 (2020-11-19): [FIRRTL] Fix LowerTypes to respect flow when connecting to instances
This fixes #255. When connecting to instances that have their bundles
flattened, we need to make sure the flattened destination is in fact a
sink, or else connect in the other direction. Incidentally, this would
have been caught by #232.
commit 4264b1a3 (2020-11-19): [FIRRTL] Fix LowerTypes to respect flow when connecting to instances (#256)

* [FIRRTL] Fix LowerTypes to respect flow when connecting to instances

This fixes #255. When connecting to instances that have their bundles
flattened, we need to make sure the flattened destination is in fact a
sink, or else connect in the other direction. Incidentally, this would
have been caught by #232.
closed 2020-11-19 by mikeurbach