llvm/circt #260: [Verilator] Invalid option : -Wpedantic

Jackwin, 2020-11-19:
Hi all,

I'm a newbie and feel interested in the CIRCT. So try to do something about CIRCT.

Clone the master branch and follow the installation steps, when running the 'ninjia  check-circt', it pops out the errors about Verilator that no invalid option: Wpedantic, shown as follows.

I also try the verilator command alone, there is still no Wpedantic option error.  It seems that Verilotor has no such option.

Are there more requirements before executing this command? Thanks ahead.

FAIL: CIRCT :: verilator/errors1.sv (150 of 155)
******************** TEST 'CIRCT :: verilator/errors1.sv' FAILED ********************
Script:
--
: 'RUN: at line 2';   not /usr/bin/verilator --cc --top-module main -Wall -Wpedantic /home/chunjie/workspace/mlir/circt/test/verilator/errors1.sv 2>&1 | /home/chunjie/workspace/mlir/circt/llvm/build/bin/FileCheck /home/chunjie/workspace/mlir/circt/test/verilator/errors1.sv
--
Exit Code: 1

Command Output (stderr):
--
/home/chunjie/workspace/mlir/circt/test/verilator/errors1.sv:10:11: error: CHECK: expected string not found in input
// CHECK: %Warning-UNUSED: {{.*}}:9:14: Signal is not used: 'rst_n'
          ^
<stdin>:1:1: note: scanning from here
%Error: Invalid Option: -Wpedantic
^

Input file: <stdin>
Check file: /home/chunjie/workspace/mlir/circt/test/verilator/errors1.sv

-dump-input=help explains the following input dump.

Input was:
<<<<<<
          1: %Error: Invalid Option: -Wpedantic
check:10     X~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ error: no match found
          2: %Error: Command Failed /usr/bin/verilator_bin --cc --top-module main -Wall -Wpedantic /home/chunjie/workspace/mlir/circt/test/verilator/errors1.sv
check:10     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

teqdruid, 2020-11-19:
What version of verilator are you using?

teqdruid, 2020-11-19:
CIRCT requires a relatively recent version of Verilator, but doesn't require Verilator. @lattner @stephenneuendorffer There are two ways we could deal with this:
1) A test to check the verilator version, which would fail if verilator is installed but not a supported version. This would at least provide a more helpful error message.
2) Detect the version of Verilator when setting up the testing, and disable the verilator feature if it doesn't find the correct version. Possibly with a "found wrong version" warning.

I think 2 is better as it's not unusual for users to have verilator installed via their Linux distro and it's more times than not the 3.x version or earlier. (4.x is supported, at least 4.034 for cosim.) 2 will, however, produce somewhat un-intuitive behavior if a user is expecting to run the verilator based tests.

I don't feel strongly either way, given that the workaround is to run the `utils/get-verilator.sh` script to install a vetted version of verilator.

stephenneuendorffer, 2020-11-19:
I think it's reasonable to check in cmake and give a message if an old version is found.  As long as the requirements are clearly layed out, probably anyone who cares about the Verilator part will be able to figure it out easily enough.

Jackwin, 2020-11-20:
Thank you for your attention and quite appreciate it @teqdruid @stephenneuendorffer 

The Verilator version is 3.916, which, I think, is the later version via the command of "apt-get install verilator". If CIRCT has a preferred version, please let me know.

BTW: For detecting the version of Verilog, it seems reasonable that the installation would tell users the version detected and the preferred version by CIRCT. So users make the choice. 

Thanks

teqdruid, 2020-11-20:
CIRCT requires at least version 4. The script under ‘utils’ will install
the preferred version into ‘ext’ and it’ll be used by the CIRCT tests.

lattner, 2020-11-20:
I'd recommend a three prong approach:

1) if verilator is not found or is explicitly disabled with a cmake flag, it is disabled.
2) if verilator is found and new enough, it is enabled.
3) if is found but too old, fail with an error that mentions the cmake flag to explicitly disable it.

teqdruid, 2020-11-20:
That sounds reasonable. Ignore my PR for the time being -- it just issues a
warning.

teqdruid, 2020-11-20:
Done

Jackwin, 2020-11-20:
@teqdruid After the Verilator is updated to the latest, everything works. Thank you.

PR #253 (2020-11-22): Improves build environment flexibility
- Adds the ability to disable or specify the paths to the two optional, external dependencies (Verilator and Capnp).
- Adds minimum version checking for Verilator
- Introduces a script to run both unit tests and integration tests in a clean environment using Docker. This should help avoid implicit local dependencies, especially in integration tests and reduce nightly test failures in the future.

Closes #260.
closed 2020-11-22 by teqdruid