llvm/circt #261: instances with flipped bundles breaks LowerTypes

darthscsi, 2020-11-19:
firrtl.circuit "foo" {
    firrtl.module @bar(%clock: !firrtl.clock, %reset: !firrtl.uint<1>) {
    }
    firrtl.module @foo(%clock: !firrtl.clock, %reset: !firrtl.uint<1>) {
      %int_bus = firrtl.instance @bar {name = "int_bus"} : !firrtl.flip<bundle<clock: clock, reset: uint<1>>>
      %0 = firrtl.subfield %int_bus("clock") : (!firrtl.flip<bundle<clock: clock, reset: uint<1>>>) -> !firrtl.flip<clock>
      firrtl.connect %0, %clock : !firrtl.flip<clock>, !firrtl.clock
      %1 = firrtl.subfield %int_bus("reset") : (!firrtl.flip<bundle<clock: clock, reset: uint<1>>>) -> !firrtl.flip<uint<1>>
      firrtl.connect %1, %reset : !firrtl.flip<uint<1>>, !firrtl.uint<1>
    }

  }

darthscsi, 2020-11-19:
@mikeurbach This is an example for my previous comment.

mikeurbach, 2020-11-19:
@darthscsi thanks I will take a look at this right now. If this is an issue right now perhaps we should remove LowerTypes from `firtool` until it is stable?

mikeurbach, 2020-11-19:
@darthscsi I guess I was confused because the spec and this [comment](https://github.com/llvm/circt/blob/master/include/circt/Dialect/FIRRTL/OpDeclarations.td#L10) led me to believe an instance's result type is always `BundleType`, with any flips on the fields themselves. However, I can see how the form you have here is handy: if every port in the instance is an input, it is convenient to flip the whole instance's result type so every field becomes an sink.

I just want to double check since I am not a FIRRTL expert: is this legal? The InstanceOp verifier obviously accepts it, and I think that is intentional.

Assuming this form is valid, the fix in LowerTypes should be simple, and I'll put up a PR for it shortly.

mikeurbach, 2020-11-19:
Furthermore, since I'm relaxing this restriction I unintentionally imposed on the result type of an instance, is there any other type I should consider here, or should it be guaranteed that the result type of an instance is either BundleType or a FlipType with an inner bundle type? I'd like to put this logic in the verifier so I can just assume it in the pass.

lattner, 2020-11-19:
> @darthscsi thanks I will take a look at this right now. If this is an issue right now perhaps we should remove LowerTypes from firtool until it is stable?

I would recommend making it off by default, but have a "-enable-lower-types" flag that turns it on in the flow.

mikeurbach, 2020-11-19:
I believe this may be what is leading to this form, so I guess this is expected: https://github.com/llvm/circt/blob/master/lib/Dialect/FIRRTL/Types.cpp#L509-L510.

> I would recommend making it off by default, but have a "-enable-lower-types" flag that turns it on in the flow.

Sounds good to me.

lattner, 2020-11-19:
Just FYI, but one difference between the FIRRTL dialect and the Scala FIRRTL implementation is that I implemented some canonicalizations for types, to enable pointer equality checking.

The key way this manifests is that `bundle(flip(T1), flip(T2))` gets canonicalized into `flip(bundle(T1, T2))`, and `flip(bundle(T1, flip(T2)))` is canonicalized into `bundle(flip(T1), T2)`.

I'm not sure if this is what is going on here, but I'm just mentioning it for visibility.

mikeurbach, 2020-11-19:
Yep I think that is what's going on here, and it makes sense now. The first canonicalization there is what is leading to instance result types that have an outer flip.

lattner, 2020-11-19:
Random thought that I haven't had time to explore: the firrtl dialect tries to model the .fir structure reasonably closely, and fir only supports a single result, so it uses a bundle for this.  We could have the firrtl.instance op produce multiple results instead, and have the fir parser handle this.

Such a modeling wouldn't be completely faithful to the scala FIRRTL implementation, but it would be way way way easier to work with.  I think it is probably the right thing to do.

mikeurbach, 2020-11-19:
I think that would be great. Using a bundle in this one special case, even in Low FIRRTL representations, is annoying, especially since MLIR has first class support for returning multiple results. If we do this, the LowerTypes pass can probably be simplified, or at least become a more idiomatic transform.

For now I am fixing LowerTypes to work with instance bundles that have been canonicalized with a flip at the outer level. I'll also make the LowerTypes pass optional and default off in firtool.

mikeurbach, 2020-11-19:
Ok, the linked PR #263 has my changes here. At the very least, we can cherry-pick the commit to disable LowerTypes in firtool, but I also included the changes I mentioned about beefing up the instance result type verification and actually fixing LowerTypes. Sorry for introducing this issue, and thanks for helping get to the bottom of it!

mikeurbach, 2020-11-20:
Closed by #263

closed 2020-11-20 by mikeurbach