llvm/circt #270: Remove `circt/llvm` submodule in favor of depending on `llvm/llvm-project`

GeorgeLyon, 2020-11-21:
I'm wondering what the effort involved in upstreaming whatever patches remain in `circt/llvm` to `llvm/llvm-project`. To be honest, I haven't been able to figure out what commits are not-yet-upstreamed because my `git` tells me there is no merge base between `llvm/llvm-project` and `circt/llvm`.

My issue is that we have a project which depends on `llvm/circt` and I needed to fix a few things in MLIR's C API. I've been looking at updating `circt/llvm` after those changes land, and the aforementioned missing-merge-base issue is making that tricky (get "refuses to merge unrelated histories" when I try to pull llvm master into circt). That got me thinking that this may be an "eat your vegetables" type of situation where the eating of vegetables involves getting patches required by `circt` actually into `llvm/llvm-project`, instead of enjoying the desserts of just keeping them in `circt/llvm`. It helps that we already have a way to test against an external checkout of llvm using the `LLVM_DIR` cmake option (for instance, if you wanted to collaborate on a patch).  I'm happy to help with this vegetable-eating, unless there is something I'm missing which makes this unfeasible (though that might be indicative of a bigger problem).

stephenneuendorffer, 2020-11-21:
George, this seems odd.  Are you sure you're looking at the right directory?

$ cd circt/llvm
$ git remote add llvm git@github.com:llvm/llvm-project.git
$ git fetch llvm
$ git merge-base llvm/master origin/master
d50149b42ec1980ca741ae626fee4fe7ff633188

So: you have to make sure you're in the submodule directory (/circt/llvm) when you pull,, rather than in /circt..

GeorgeLyon, 2020-11-21:
OK, so this is interesting:
```
$ git rev-parse origin/master                                  
d50149b42ec1980ca741ae626fee4fe7ff633188
$ git merge-base upstream/master d50149b42ec1980ca741ae626fee4fe7ff633188
d50149b42ec1980ca741ae626fee4fe7ff633188
$ git rev-parse HEAD
2e36e0dad52b07f0e856f939d530d47bbe8a74ac
$ git merge-base upstream/master HEAD
<no output>
$ git rev-parse origin/master HEAD
<no output>
```

I apologize if I am misinterpreting `git`, but it seems to me that:
1. There are no commits in `circt/llvm master` which are not in `llvm/llvm-project master`
2. The commit currently referenced by `llvm/circt` is not on master, and does not share a history with master.

The only reason I can think for this to be the case is if history was rewritten at some point.

stephenneuendorffer, 2020-11-21:
Yes, you're right about #2.  It seems that `circt/llvm master` was not updated.  This isn't really necessary.  However, I do see that this commit exists in llvm/llvm-project master: https://github.com/llvm/llvm-project/commit/2e36e0dad52b07f0e856f939d530d47bbe8a74ac

I believe that there are no commits in circt/llvm that are not upstream.

I suspect you might need to `git submodule update` or `cd circt/llvm; git fetch`?

darthscsi, 2020-11-22:
> master: llvm/llvm-project@2e36e0d

updating what the submodule is pointing to is something you can push
without a pr as long as it doesn't break anything.

mikeurbach, 2020-11-22:
> I believe that there are no commits in circt/llvm that are not upstream.

Not from me. I believe @teqdruid is the only other person who has been pushing changes to MLIR recently, and I think all his patches have landed as well.

> The commit currently referenced by llvm/circt is not on master

Do you mean your local `master` branch, or the `master` branch on the GitHub repo? I believe it is in the history of the master branch on GitHub: https://github.com/circt/llvm/commit/2e36e0dad52b07f0e856f939d530d47bbe8a74ac. If you have the submodule inside circt/llvm set up with your local master branch tracking the master branch on that GitHub repo, Stephen's suggestion to `cd circt/llvm; git fetch` should bring that change down locally. You may also need to then `git pull` on `master` to fast forward your local branch.

GeorgeLyon, 2020-11-22:
I've worked through the merge-base issue and closing this for now as there didn't seem to be much enthusiasm for getting rid of `circt/llvm`

closed 2020-11-22 by GeorgeLyon