llvm/circt #277: [ODS] Autogenerated "getType()" member could be more specific

youngar, 2020-11-23:
Most ops in the RTL dialect require the result type to be `AnySignlessInteger`.  When getting the result type of an operation using `getType()`, a `mlir::Type` is returned which is usually immediately cast to `mlir::IntegerType`.  This leads to a lot of code which looks like the following:

```c++
auto resultTy = getType().cast<IntegerType>();
```

It would be nice if `getType()` returned the correct type for us in situations where it could be inferred, and the cast could be avoided.

```c++
auto resultTy = getType();
```

Currently, the `getType()` call comes from the `[OneResult]` trait on the operation.

This issue is mostly for tracking the local CIRCT solution to the problem.  When I have time, I will put something on the mailing list.

### Some background

Originally brought up in https://github.com/llvm/circt/pull/269#pullrequestreview-535978657

> Do you know if there is a way to make `getType()` on `ZextOp` return an `IntegerType`? I've always wondered this, I'm not sure if it is a missing ODS feature or if it is there but I don't know how to use it.

---

Chris filed an LLVM Bug for this: https://bugs.llvm.org/show_bug.cgi?id=46698

> For an operation with a single result:
> 
>  let results = (outs AnySignlessInteger:$result);
> 
> ODS is currently synthesizing a `Type getType()` member.  It would be helpful if that was `IntegerType getType()` to reduce casts in client code.

---

Semi-related improvement to `[FirstAttrDerivedResultType]` and `[SameOperandsAndResultType]`:
```
  // FIXME(QoI): Instead of requiring "rtl.constant (42: i8) : i8", we should
  // just use "rtl.constant 42: i8".  This can be done with a custom printer and
  // parser, but would be better to be autoderived from the
  // FirstAttrDerivedResultType trait.  This should be an MLIR core improvement.
  let assemblyFormat = [{
     `(` $value `)` attr-dict `:` type($result)
  }];
```

lattner, 2020-11-24:
Cool, thanks for filing this Andrew.  I think it would be best to work with the upstream MLIR community on this.  Once it converges and lands there we can merge it downstream into the CIRCT clone of LLVM.  I think that @River707 will be the best reviewer for this.

lattner, 2020-12-24:
There is a patch inflight over here, I'm going to close this since it is tracked on the MLIR side of things:
https://reviews.llvm.org/D93790

closed 2020-12-24 by lattner