llvm/circt #284: [RTL] Sequential logic

teqdruid, 2020-11-24:
Add sequential logic elements like clocks and registers. We haven't agreed on how this should be modeled yet, so this issue should be a place for discussion.

mikeurbach, 2020-11-24:
I've been wondering when this would come up, and I'm happy to join the discussion. I recently added some sequential logic to one of the Handshake lowerings to FIRRTL, which got me thinking about this. I can go directly from the FIRRTL to EmitVerilog and get the System Verilog output I would expect. I was thinking about how this could work with the existing lowering from FIRRTL to RTL... I will try to flesh out my ideas and bring them here.

teqdruid, 2020-11-24:
This conversation got started in #282. My proposal can be found there.

mikeurbach, 2020-12-28:
I see a `rtl.reg` op was recently added here: https://github.com/llvm/circt/commit/1caf7ddf825799fe6fbc6f09ed5e8e233aa345b1.

It looks like the op declares a register, which can be connected to with `rtl.connect` to represent a procedural continuous assignment with the System Verilog `assign` keyword. I think this makes sense, but I'm wondering how to represent sequential logic using values declared by the new `rtl.reg` op.

There is already a `sv.alwaysat_posedge` op, which gets us pretty close, for example:

```mlir
module {
  rtl.module @test(%input : i8, %clk : i1) -> (%output : i8) {
    %myReg = rtl.reg : !rtl.inout<i8>
    sv.alwaysat_posedge %clk {
      rtl.connect %myReg, %input : i8
    }
    %0 = rtl.read_inout %myReg : !rtl.inout<i8>
    rtl.output %0 : i8
  }
}                                                                                                                                                                                                 
```

Will emit:

```mlir
module test(
  input  [7:0] input,
  input        clk,
  output [7:0] output);

  reg [7:0] myReg;	// <stdin>:3:14

  always @(posedge clk) begin	// <stdin>:4:5
    assign myReg = input_0;	// <stdin>:5:7
  end // always @(posedge)
  assign output = myReg;	// <stdin>:7:10, :8:5
endmodule
```

I have a few thoughts about the above:

* I think the combination of `rtl.reg`, `rtl.read_inout`, and `sv.alwaysat_posedge` is a good start, but does this pattern make sense?
* If we are going roughly in this direction, we can probably generalize the `sv.alwaysat_posedge` op (which there is already a TODO for) to handle multiple signals (e.g. a negedge reset)
  * Or, come up with a new op like `sv.always_ff` to more closely model how sequential logic should be specified in System Verilog (is `always_ff` the preferred way to specify sequential logic in System Verilog?)
* The usage of `rtl.connect` is sort of just a placeholder, since I don't think we have an op with the desired semantics of a non-blocking assignment. Ultimately, I think we'd want something that would emit `myReg <= input` here, not `assign myReg = input`. I think this could go a number of ways, and I'm interested to hear what others think:
  * Come up with a new `rtl` op that models non-blocking assignment
  * Add an attribute to `rtl.connect` to denote whether the connection is a continuous assignment with `assign` or a non-blocking assignment. Similar to the previous point, just a different way to represent the same information.
  * Add an op to the SV dialect for non-blocking assignment

The previous point is what I'm most interested to hear some feedback about. The RTL dialect currently has the notion of a "connect" op which is interpreted as a System Verilog `assign` when exporting System Verilog. Does it make sense to have an op for the non-blocking assignment `<=` from Verilog/System Verilog in the RTL dialect? Is this a concept that makes sense in HDLs in general? I'm thinking about an RTL module that should be emitted to VHDL or System Verilog. I'm also thinking about lowering RTL modules to LLHD for simulations.

It seems like having the ability to represent non-blocking assignment in the RTL dialect would be a nice way to share this representation in one place, but if these semantics are closely tied to Verilog / System Verilog it might not make sense.

teqdruid, 2020-12-29:
Blocking vs non-blocking vs continuous and `wire` vs `reg` are very SystemVerilog views of the world and I don't think belong in the RTL dialect. I also don't think there should be an `rtl.connect` op. Nor should `rtl.wire` or the new `rtl.reg` exist, IMO. They should live in the SV dialect.

My view on  SV vs RTL dialects:
- The SV dialect is a 'structural' dialect, meaning that one cannot make much sense of data flow using standard SSA dataflow analysis. Most of the ops are side-effectful, such as `sv.connect` (and `rtl.connect`). It also has all the SystemVerilog quirks such as `wire`, `reg`, and `logic`.
- The RTL dialect is a real SSA dialect containing as few side-effectful ops as possible, using values instead. This enables the use of dataflow analysis directly on the SSA.

As such, I think the SV parts of your proposal are fine but I am loathe to introduce those concepts to the RTL dialect (vis-a-vis the `rtl.connect` attribute. I also think it's worth a longer discussion on how to model state in the RTL dialect. It's not clear to me whether or not https://github.com/llvm/circt/commit/1caf7ddf825799fe6fbc6f09ed5e8e233aa345b1 was attempting to model registers or just copying the notion of a SystemVerilog `reg`.

teqdruid, 2020-12-29:
It occurs to me that nobody's written up a clear description of what the SV dialect should be and what the RTL dialect should be. (Am I wrong and just not finding it?) I think this would come in handy.

mikeurbach, 2020-12-29:
Not that I know of, and I agree that would be very helpful. I agree with your general view about the roles of the RTL and SV dialects. I guess I didn't realize how specific the non-blocking assignment was to Verilog land.

teqdruid, 2020-12-29:
There's nothing fundamental to hardware about blocking vs. non-blocking -- it's just a style thing (IMO) of SystemVerilog.

lattner, 2020-12-30:
Just to clarify here, I have no specific need for "reg" to be in the RTL dialect, it would be fine to move it to the SV dialect if that helps.  My understand is incomplete, but I think it is just a different spelling for humans and tools that want a hint about its behavior.  Putting it in SV makes perfect sense to me.

> I think the combination of rtl.reg, rtl.read_inout, and sv.alwaysat_posedge is a good start, but does this pattern make sense?

I think the later two do, but I agree that `reg` could be moved to SV, feel free to do so!

> If we are going roughly in this direction, we can probably generalize the sv.alwaysat_posedge op (which there is already a TODO for) to handle multiple signals (e.g. a negedge reset)

Yes definitely, I happened to do that today because I needed it for RegInit lowering!

> Or, come up with a new op like sv.always_ff to more closely model how sequential logic should be specified in System Verilog (is always_ff the preferred way to specify sequential logic in System Verilog?)

Also having an `sv.always_ff` would make sense if we have a client for it.

> The usage of rtl.connect is sort of just a placeholder, since I don't think we have an op with the desired semantics of a non-blocking assignment.

We actually have three things now:

1) rtl.connect with the semantics that "this is wired to that", it is printed as an `assign x = y` statement.
2) sv.bpassign which is a blocking procedural assignment, printing as `x = y`
3) sv.passign which is a nonblocking procedural assignment, printing as `x <= y`

My understanding is that the later two only make sense in always and other procedural sorts of things.  Such statements are really just sugar for muxes etc, which is why I shoved them into the SV namespace.

>There's nothing fundamental to hardware about blocking vs. non-blocking -- it's just a style thing (IMO) of SystemVerilog. 

I don't think that's true, they have different semantics when it comes to simulation.

lattner, 2020-12-30:
Keep in mind that the procedural assignment statements tie into the weird verilog behavior that is pseudo-imperative with 'last connect wins' semantics.  This is where you get things like:

x <= y
if (thing)
  x <= z

which is sugar for something like `assign x = thing ? z : y;`

teqdruid, 2020-12-30:
> Just to clarify here, I have no specific need for "reg" to be in the RTL dialect, it would be fine to move it to the SV dialect if that helps. My understand is incomplete, but I think it is just a different spelling for humans and tools that want a hint about its behavior.

My understanding is the same.

>>There's nothing fundamental to hardware about blocking vs. non-blocking -- it's just a style thing (IMO) of SystemVerilog.
> I don't think that's true, they have different semantics when it comes to simulation.

By "style" I was being hyperbolic for "quirk". Yes, they are functionally different in SV but if used properly model the same hardware.

lattner, 2021-06-20:
Does this bug serve a purpose?  I don't think we need to keep this open to track this

teqdruid, 2021-06-21:
I agree. Thanks for the gardening!

closed 2021-06-21 by teqdruid