llvm/circt #286: Unable to lower to FIRRTL from handshake using -lower-handshake-to-firrtl with internally defined memref

JuanEsco063, 2020-11-24:
I have the following sample code in standard:

func @test (%arg0: index) ->f32 {
%0 = alloc() : memref<10xf32>
%c0 = constant 0 : index
%c10 = constant 10.0 : f32
store %c10, %0[%c0] : memref<10xf32>
%5 = load %0[%c0] : memref<10xf32>
return %5 : f32
}

Which I can lower to Handshake dialect, obtaining:

module {
  handshake.func @test(%arg0: index, %arg1: none, ...) -> (f32, none) {
    %0:3 = "handshake.memory"(%9#0, %9#1, %addressResults) {id = 0 : i32, ld_count = 1 : i32, lsq = false, st_count = 1 : i32, type = memref<10xf32>} : (f32, index, index) -> (f32, none, none)
    %1:2 = "handshake.fork"(%0#1) {control = false} : (none) -> (none, none)
    %2 = "handshake.merge"(%arg0) : (index) -> index
    "handshake.sink"(%2) : (index) -> ()
    %3:3 = "handshake.fork"(%arg1) {control = true} : (none) -> (none, none, none)
    %4:3 = "handshake.fork"(%3#2) {control = true} : (none) -> (none, none, none)
    %5 = "handshake.join"(%4#2, %1#1, %0#2) {control = true} : (none, none, none) -> none
    %6 = "handshake.constant"(%4#1) {value = 0 : index} : (none) -> index
    %7:2 = "handshake.fork"(%6) {control = false} : (index) -> (index, index)
    %8 = "handshake.constant"(%4#0) {value = 1.000000e+01 : f32} : (none) -> f32
    %9:2 = "handshake.store"(%8, %7#0, %3#1) : (f32, index, none) -> (f32, index)
    %10 = "handshake.join"(%3#0, %1#0) {control = true} : (none, none) -> none
    %11, %addressResults = "handshake.load"(%7#1, %0#0, %10) : (index, f32, none) -> (f32, index)
    handshake.return %11, %5 : f32, none
  }
}

But when I try to lower that to FIRRTL from it, I get the following errors:

circt-opt: ../llvm/mlir/include/mlir/IR/Types.h:296: bool mlir::Type::isa() const [U = circt::firrtl::BundleType]: Assertion `impl && "isa<> used on a null type."' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0. Program arguments: circt-opt -lower-handshake-to-firrtl test_1_HS.mlir 
1. In-Flight Diagnostics:
    loc("test_1_HS.mlir":2:3): error: Unsupported data type. Supported data types: integer (signed, unsigned, signless), index, none.
    loc("test_1_HS.mlir":3:12): error: Unsupported data type. Supported data types: integer (signed, unsigned, signless), index, none.
    loc("test_1_HS.mlir":3:12): error: Unsupported data type. Supported data types: integer (signed, unsigned, signless), index, none.

Which points to the arguments of the test function and the arguments of the handshake memory element.

mikeurbach, 2020-11-24:
Thanks for raising this issue. I saw the same, but didn't get around to filing it.

I think there are two things here.

1. The HandshakeToFIRRTL pass should use the relatively new feature `getDependentDialects` to declare its dependency on the FIRRTL dialect. I think that will resolve the crash.

2. The in-flight diagnostics at the time of the crash point to the real issue: Handshake simply doesn't support float types. Those diagnostics are being emitted from this check: https://github.com/llvm/circt/blob/master/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp#L251. In general, I don't think there is any support for float types in the layers below Handshake either.

I think 1. is going to be relatively straightforward to address, but 2. is a much larger discussion :)

JuanEsco063, 2020-11-24:
Thank you for the quick reply!

I see. At the stage of the project I am right now we can just switch to i32 type to avoid the float issue. Changing the data type to i32 gets rid of those errors but now I get this new one:

circt-opt: ../llvm/llvm/include/llvm/ADT/SmallVector.h:183: llvm::SmallVectorTemplateCommon::reference llvm::SmallVectorTemplateCommon<mlir::Value, void>::operator[](llvm::SmallVectorTemplateCommon::size_type) [T = mlir::Value]: Assertion `idx < size()' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0. Program arguments: circt-opt -lower-handshake-to-firrtl test_1_HS.mlir 
1. In-Flight Diagnostics:
    loc("test_1_HS.mlir":3:12): error: Usupported operation type

Again, pointing to the Handshake memory element. Is this still related to the getDependentDialects  feature you mentioned? If so, where can I find information on how to make sure this dependency is correctly set?

Thanks beforehand!

mikeurbach, 2020-11-24:
Hmm, I am less sure of the cause of that crash or where that In-Flight Diagnostic is generated. I've seen similar errors recently, but I'm not sure if they were related. Do you mind sharing the full backtrace here?

I don't know if fixing `getDependentDialects` for this pass will resolve the issue, but here is an example of what that looks like: https://github.com/llvm/circt/commit/b117102930421fb1da9562ceb59ad2f367a5f44b#diff-27a72acf52128f4feebd056aab09b7d92ef0737b681f3fd1bdce72adbfa4a7c3

JuanEsco063, 2020-11-24:
Sure! I'll do it right now. Thank you for your help!

JuanEsco063, 2020-11-24:
circt-opt: ../llvm/llvm/include/llvm/ADT/SmallVector.h:183: llvm::SmallVectorTemplateCommon::reference llvm::SmallVectorTemplateCommon<mlir::Value, void>::operator[](llvm::SmallVectorTemplateCommon::size_type) [T = mlir::Value]: Assertion `idx < size()' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0. Program arguments: circt-opt -lower-handshake-to-firrtl test_1_HS.mlir 
1. In-Flight Diagnostics:
    loc("test_1_HS.mlir":3:12): error: Usupported operation type
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
circt-opt[0x4c893a]
circt-opt[0x4c8b0b]
circt-opt[0x4c70fb]
circt-opt[0x4c925d]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x153c0)[0x7f80dc8323c0]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f80dc2e318b]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f80dc2c2859]
/lib/x86_64-linux-gnu/libc.so.6(+0x25729)[0x7f80dc2c2729]
/lib/x86_64-linux-gnu/libc.so.6(+0x36f36)[0x7f80dc2d3f36]
circt-opt[0x57852c]
circt-opt[0x6f9a5b]
circt-opt[0x6f978b]
circt-opt[0x6f96e6]
circt-opt[0x6f95a5]
circt-opt[0x6f8ed1]
circt-opt[0x6f89b1]
circt-opt[0x6f8231]
circt-opt[0x6f7fc1]
circt-opt[0x6f7ca1]
circt-opt[0x6f7b11]
circt-opt[0x6f663d]
circt-opt[0x7030fc]
circt-opt[0x703340]
circt-opt[0x702d97]
circt-opt[0xc3ee7c]
circt-opt[0xc99f61]
circt-opt[0xc99e5e]
circt-opt[0xc47c60]
circt-opt[0xc47310]
circt-opt[0xc46912]
circt-opt[0xc408ab]
circt-opt[0xc405bb]
circt-opt[0xc40a32]
circt-opt[0x6f4cf4]
circt-opt[0x107cd25]
circt-opt[0x107d27d]
circt-opt[0x1084b6d]
circt-opt[0x10844d2]
circt-opt[0x1084460]
circt-opt[0x107e3e4]
circt-opt[0x107db84]
circt-opt[0x107cfed]
circt-opt[0x107cd13]
circt-opt[0x107d27d]
circt-opt[0x107f06d]
circt-opt[0x960416]
circt-opt[0x95f6f0]
circt-opt[0x95f546]
circt-opt[0x40c741]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3)[0x7f80dc2c40b3]
circt-opt[0x40c19e]
Aborted (core dumped)

mikeurbach, 2020-11-24:
Hmm, seems like that's a release build of CIRCT? I can't really see what's going on. Maybe try building CIRCT with `-DCMAKE_BUILD_TYPE=Debug` or `-DCMAKE_BUILD_TYPE=RelWithDebInfo`?

The backtrace should have file and line numbers, and function names, not just `circt-opt[some address]`.

JuanEsco063, 2020-11-24:
Probably. I just followed the instructions on the CIRCT repo to build it. Let me try it again with those options

hanchenye, 2020-11-24:
Hi Juan, the diagnostics actually point out the issue, currently the HandshakeToFIRRTL lowering doesn't support handshake load/store operations.

Both the Handshake and StdExpr dispatcher failed to handle the handshake load/store, and errors are emitted here:
https://github.com/llvm/circt/blob/0eba89a76aae693c9101ea9d7b56af5f827f6ebb/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp#L1239

Thanks for raising this issue! I'll be happy to take a look and work on the lowering of load/store.

mikeurbach, 2020-11-24:
Ah, thanks @hanchenye. I thought I could grep for `Unsupported operation type`, but I wasn't careful, and it looks like that's not how it's spelled in the diagnostic. Meta note: how do you make such a nice link?

So it seems like this issue is really a feature request to lower Handhsake load/store? That sounds interesting, and I would make use of such functionality as well!

hanchenye, 2020-11-24:
Auh, @mikeurbach when you click the line number, there's a "copy permalink" option. I guess only a link referring to a hash id will have this pretty look :D

JuanEsco063, 2020-11-25:
Thank you for your help @hanchenye and @mikeurbach! I re-build the repo and made sure to have the debug option enabled and still only get the hex code instead of file and line number. But @hanchenye comment pretty much answers my issue.

closed 2020-11-25 by JuanEsco063