llvm/circt #297: [CMAKE] CIRCT targets not exported

youngar, 2020-12-02:
When using CIRCT from other cmake projects, another project should be able to use CIRCT like so:

```cmake
project(my-circuit-project
find_package(CIRCT REQUIRED CONFIG)
add_executable(my-tool)
target_link_libraries(my-tool PUBLIC MLIRFIRRTL)
```
The `find_package()` command loads `CIRCTConfig.cmake`, which in turn loads `CIRCTTargets.cmake`. `CIRCTTargets` is where all targets exported from CIRCT must be listed. `CIRCTTargets.cmake` must be populated somewhat manually. In MLIR and CIRCT, we export every target in `${MLIR_EXPORTS}` and `${CIRCT_EXPORTS}`, respectively:
```cmake
# Generate a list of CMake library targets so that other CMake projects can
# link against them. LLVM calls its version of this file LLVMExports.cmake, but
# the usual CMake convention seems to be ${Project}Targets.cmake.
get_property(CIRCT_EXPORTS GLOBAL PROPERTY CIRCT_EXPORTS)
export(TARGETS ${CIRCT_EXPORTS} FILE ${circt_cmake_builddir}/CIRCTTargets.cmake)
```

In CIRCT, since we are using `add_mlir_library()` and friends, the `${CIRCT_EXPORTS}` is actually empty, while `${MLIR_EXPORTS}` contains all our targets. This means that `CIRCTTargets.cmake` is empty, and `find_package(CIRCT)` does not allow anyone to link against CIRCT libraries. We can change `CIRCTTargets.cmake` to export `MLIR_EXPORTS` instead, but that may be problematic if we try to build in the MLIR mono-repo.  (I'm guessing we will start exporting all MLIR targets, and have problems with double-exporting targets from the build tree).

A better long term solution might be to add `add_circt_library()`.  These functions can start off as thin wrappers around `add_mlir_library()` which also populate `CIRCT_EXPORTS`.  Eventually, we might customize them more, such as adding the ability to run `clang-tidy` on just CIRCT libraries.  This might make it easier to transition to the llvm mono-repo in the long run.

stephenneuendorffer, 2020-12-02:
Andrew, there's actually a bunch of stuff that LLVM-like projects need.  I started to consolidate this at one point, but haven't returned to it.  See: https://reviews.llvm.org/rGe9b87f43bde8b5f0d8a79c5884fdce639b12e0ca
Even without going that far, I agree that this is a good thing to do.

youngar, 2020-12-23:
CIRCT Exports should be working as of #306

closed 2020-12-23 by youngar