llvm/circt #299: [FIRRTL] reconsider firrtl.node

lattner, 2020-12-04:
Andrew Young mentioned today that the existence of `firrtl.node` is a bit weird given MLIR's SSA based design (it makes more sense for the scala implementation of FIRRTL, which doesn't have an efficient "RAUW" operation).

We should figure out what to do about this, some possibilities:

1) We could have the parser zap nodes when reading fir files, defining them out of our internal IR.
1a) We could have the parser turn them into firrtl.wire ops in some cases, e.g. when there exists a name.
2) We could represent them, but always aggressively canonicalize them away.

The big question to me is "can the scala implementation of firrtl put annotations on firrtl.node objects?  If not, then I don't think there is any value in preserving them.  If so, I think we can promote them to wires in the case where they have an annotation.  Either way it would be great to have them go away.

There is a related question that will come up at some point, which is "do we want something like firrtl.node that we can lower wires into while preserving names".  Such a concept would look much like firrtl.node, but would have to be modeled as having side effects.  I'm not aware of any current need in the FIRRTL flow to lower from wires, so I'd recommend separating this off and deferring it until it comes up.

jackkoenig, 2020-12-04:
> The big question to me is "can the scala implementation of firrtl put annotations on firrtl.node objects? 

Yes, they are named and so can be annotated.

>  If so, I think we can promote them to wires in the case where they have an annotation. Either way it would be great to have them go away.

Promoting them to wires is equivalent and makes sense for the use case of annotated nodes

> I'm not aware of any current need in the FIRRTL flow to lower from wires, so I'd recommend separating this off and deferring it until it comes up.

In the Scala FIRRTL compiler at least, converting from wires to nodes makes all combinational logic feed forwards which made it much easier to implement optimizations. Assuming your wires are represented in SSA you probably already get this benefit. If CIRCT firrtl wires are *not* SSA, then depending on how your optimizations are implemented you may see improvements like we did.

youngar, 2021-05-14:
I think we can close this now.  We have overall a much better understanding of when entities can be removed, and our support for this is evolving.  We have a [canonicalization](https://github.com/llvm/circt/pull/904/files) which will optimize away certain nodes, and [support for eliminating nameless nodes](https://github.com/llvm/circt/commit/2b9ca28432350bf6c58f7ac1855ba5db49d88501) in the parser.

closed 2021-05-14 by youngar

lattner, 2021-05-14:
+1 thanks