llvm/circt #311: [FIRRTL] (or) operation with duplicate input

drom, 2020-12-10:
`firtool` with the following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_r: SInt<38>
    output tmp874: UInt<38>
    tmp874 <= or(inp_r, inp_r)
```

Generates this error:

```mlir
drom@linux-sydi:~/work/github/drom/circt-fuzzer> ../../llvm/circt/build/bin/firtool top_mod.fir
top_mod.fir:6:12: error: type mismatch between destination '!firrtl.uint<38>' and source '!firrtl.sint<38>'
    tmp874 <= or(inp_r, inp_r)
           ^
top_mod.fir:6:12: note: see current operation: "firrtl.connect"(%tmp874, %inp_r) : (!firrtl.flip<uint<38>>, !firrtl.sint<38>) -> ()
```

Is this expected?

lattner, 2020-12-10:
Great catch, this is definitely a bug.  It looks like the folder from "or(x,x) -> x" isn't adding a cast from SInt operands to UInt operands.  The bug is in `OrPrimOp::fold`: This fold (and the similar ones for and/xor) can only apply to inputs who are unsigned already.  The fancier "generate a cast" version can't be implemented in the `fold` method.

lattner, 2020-12-10:
Also, really awesome reduction!!

drom, 2020-12-10:
first results using new fuzzer ;)
https://github.com/drom/circt-fuzzer

drom, 2020-12-10:
strangely `and`, `xor` are fine. only `or` is broken.

lattner, 2020-12-10:
Yeah, xor returns zero, `and` has already been fixed:

```
  /// and(x, x) -> x
  if (lhs() == rhs() && rhs().getType() == getType())
    return rhs();
```

Or just needs the `&& types match` piece

lattner, 2020-12-10:
> first results using new fuzzer ;)
> https://github.com/drom/circt-fuzzer

super awesome!!

youngar, 2020-12-10:
Fixed here: https://github.com/llvm/circt/pull/312/files

closed 2020-12-11 by youngar