llvm/circt #319: [FIRRTL] (cat) 0-bit width

drom, 2020-12-11:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_0: UInt<0>
    input inp_1: UInt<5>
    output out_0: UInt<5>
    out_0 <= cat(inp_1, inp_0)
```

Compiled with firtool --lower-to-rtl produces this verilog:

```verilog
module top_mod(
  input        inp_0,
  input  [4:0] inp_1,
  output [4:0] out_0);

  assign out_0 = {inp_1, inp_0};	// top_mod.fir:2:3, :6:14
endmodule
```

That has the following error:

```bash
%Warning-WIDTH: top_mod.v:6:16: Operator ASSIGNW expects 5 bits on the Assign RHS, but Assign RHS's REPLICATE generates 6 bits.
                              : ... In instance top_mod
    6 |   assign out_0 = {inp_1, inp_0};  
      |                ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
```

Expected Verilog output is:

```verilog
module top_mod(
  input  [4:0] inp_1,
  output [4:0] out_0);

  assign out_0 = inp_1;	// top_mod.fir:2:3, :6:14
endmodule
```

lattner, 2020-12-13:
Is there a way to generate a zero bit value in Verilog?  Maybe an empty concat like `{}`?  For example, is this a valid translation of this file:

```
module top_mod(
  input  [4:0] inp_1,
  output [4:0] out_0);

  assign out_0 = {inp_1, {}};	// top_mod.fir:2:3, :6:14
endmodule
```

I understand that we want to avoid zero width values in general and stomp them out, but if verilog allows them then this allows us to have zero bit values more easily in the RTL dialect.

-Chris

drom, 2020-12-15:
@lattner I am afraid, that the synthesizable RTL subset of Verilog would not be happy with `{}` in general.

[image]

How about emitting a comment?

```verilog
module top_mod(
  input  [4:0] inp_1,
  output [4:0] out_0);

  assign out_0 = {inp_1 /*, inp_0 */};	// top_mod.fir:2:3, :6:14
endmodule
```

lattner, 2020-12-27:
Ok, a comment isn't going to work for us in general.  We need something that can be used as an expression.  I think this means that the LowerToRTL pass has to drop zero width things.  This is fine way to handle the problem.

lattner, 2021-01-10:
Fixed in 4d89644, but there is a lot more to go on zero bit lowering.

closed 2021-01-10 by lattner