llvm/circt #324: [FIRRTL] How does Analog type work with instances?

lattner, 2020-12-14:
Analog types are implicitly "inout" sort of things that can only be casted to integer and clock when they are used as an input.  

A few questions:

1) Can they be used as output ports?  If so, are they of type `flip<Analog<42>>` or just `Analog<42>`?
2) When dealing with an instance of a module that takes Analog inputs, do we get a bundle result with a flipped analog type?

@seldridge do you know how this works?

seldridge, 2020-12-14:
@lattner wrote:
> Analog types are implicitly "inout" sort of things that can only be casted to integer and clock when they are used as an input.

I think this is the correct formulation. Note that [the SFC is more relaxed and lets almost anything appear as a righ-hand-side value](https://github.com/chipsalliance/firrtl/issues/2013). This will be tightened up at CHIRRTL time.

@lattner wrote:
> 1. Can they be used as output ports? If so, are they of type flip<Analog<42>> or just Analog<42>?

Yes, they can be declared as output or input.

The SFC stores these both as type `Analog<42>` since it treats the root flow (effectively flip) as independent of the type. Since the MFC doesn't have input/output and relies on flip, I would suggest storing these as flip and non-flip.

Note: the SFC does have the notion of a "module type" which will return a bundle type for a module. In this view you get flips and non-flips in the bundle type.

@lattner wrote:
> 2. When dealing with an instance of a module that takes Analog inputs, do we get a bundle result with a flipped analog type?

Yes. I'd treat this the same as the other ground types.

As an alternative, we could consider dropping flips from analog entirely. I haven't thought through the implications of this.

lattner, 2021-01-03:
I think that you're right -- throwing away flips of analog types makes everything simpler.  I implemented this in 1735422.  I think this thoroughly defeats the classical notions of flows :-)

closed 2021-01-03 by lattner