llvm/circt #326: [FIRRTL] (shr) core dump

drom, 2020-12-15:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    output tmp57: SInt<1>
    tmp57 <= shr(SInt<7>(29), 47)
```

Compiled with `firtool --lower-to-rtl` triggers this error:

```bash
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/APInt.h:971: void llvm::APInt::ashrInPlace(unsigned int): Assertion `ShiftAmt <= BitWidth && "Invalid shift amount"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool --lower-to-rtl top_mod.fir --verilog -o=top_mod.v
 #0 0x00000000004bd25f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:563:22
 #1 0x00000000004bd316 PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:630:1
 #2 0x00000000004bb332 llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004bccb2 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:405:1
 #4 0x00007fd442b431d0 __restore_rt (/lib64/libpthread.so.0+0x141d0)
 #5 0x00007fd442602a65 raise (/lib64/libc.so.6+0x3da65)
 #6 0x00007fd4425eb864 abort (/lib64/libc.so.6+0x26864)
 #7 0x00007fd4425eb749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x00007fd4425faf96 (/lib64/libc.so.6+0x35f96)
 #9 0x00000000004e868c llvm::APInt::ashrInPlace(unsigned int) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/APInt.h:972:21
#10 0x000000000076f5b9 llvm::APInt::ashr(unsigned int) const /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/APInt.h:966:12
#11 0x000000000076aa33 circt::firrtl::ShrPrimOp::fold(llvm::ArrayRef<mlir::Attribute>) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/OpFolds.cpp:374:37
#12 0x0000000000754d13 mlir::LogicalResult mlir::Op<circt::firrtl::ShrPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::MemoryEffectOpInterface::Trait>::foldSingleResultHook<circt::firrtl::ShrPrimOp>(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/OpDefinition.h:1641:53
#13 0x000000000068c45c mlir::AbstractOperation::foldHook(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:138:3
#14 0x00000000006860ba mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:566:30
#15 0x0000000000733897 mlir::OperationFolder::tryToFold(mlir::OpBuilder&, mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::function_ref<void (mlir::Operation*)>) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:186:13
#16 0x00000000007331de mlir::OperationFolder::tryToFold(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, llvm::function_ref<void (mlir::Operation*)>, bool*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:89:13
#17 0x0000000000716c23 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>, int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:191:21
#18 0x0000000000716f4e mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:258:35
#19 0x0000000000716e3a mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:238:1
#20 0x0000000000707e93 (anonymous namespace)::Canonicalizer::runOnOperation() /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:35:71
#21 0x000000000084f5b3 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:368:36
#22 0x000000000084f76b mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:403:15
#23 0x000000000085167b mlir::PassManager::run(mlir::ModuleOp) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:808:73
#24 0x000000000040952d processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:128:15
#25 0x00000000004099a8 main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:181:13
#26 0x00007fd4425ed152 __libc_start_main (/lib64/libc.so.6+0x28152)
#27 0x00000000004090ee _start /home/abuild/rpmbuild/BUILD/glibc-2.32/csu/../sysdeps/x86_64/start.S:122:0
Aborted (core dumped)
```

lattner, 2020-12-15:
Working on this.

lattner, 2020-12-15:
Fixed in 38cc3e8

closed 2020-12-15 by lattner