llvm/circt #328: [FIRRTL] (mul) 0-width

drom, 2020-12-15:
Maybe related to #319

The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_g: SInt<0>
    input inp_13: SInt<383>
    output tmp75: SInt<383>
    tmp75 <= mul(inp_13, inp_g)
```
Compiled with `firtool --lower-to-rtl` produces this verilog:

```verilog
module top_mod(
  input          inp_g,
  input  [382:0] inp_13,
  output [382:0] tmp75);

  assign tmp75 = inp_13 * {{383{inp_g[4294967295]}}, inp_g};	// top_mod.fir:2:3, :6:14
endmodule
```

That has the following error:

```bash
%Error: top_mod.v:6:38: Illegal bit or array select; type does not have a bit range, or bad dimension: data type is 'logic'
                      : ... In instance top_mod
    6 |   assign tmp75 = inp_13 * {{383{inp_g[4294967295]}}, inp_g};  
      |                                      ^
```

lattner, 2020-12-15:
Thanks!  I'd recommend taking zero bit support out of your fuzzer until we have a chance to determine a design point for supporting them correctly.

lattner, 2021-01-10:
Out of curiosity, how does SFC handle this?  Are zero bit fields sign/zero extended to zero values?  @seldridge do you know?

seldridge, 2021-01-10:
It depends... the default for an expression is to become a `UInt<1>(0)`. However, there are a lot of special cases.

Two examples:

- `cat` will drop zero-width arguments (it's more complicated than just this, though)
- `andr` becomes `UInt<1>(1)` (for the reason that an and reduction is defined as a fold with an initial value of one)

Zero widths are removed in the `ZeroWidth` transform, with the code related to those two examples [here](https://github.com/chipsalliance/firrtl/blob/15013df6f6ac2dafeb35d7ed15cf95c7ac8a5bef/src/main/scala/firrtl/passes/ZeroWidth.scala#L128).

Note: there is also a [`ZeroLengthVecs` transform](https://github.com/chipsalliance/firrtl/blob/15013df6f6ac2dafeb35d7ed15cf95c7ac8a5bef/src/main/scala/firrtl/passes/ZeroLengthVecs.scala) that runs earlier than `ZeroWidth` which will handle zero length vectors.

`mul` uses the default case and looks like the following.

High:
```
circuit Foo :
  module Foo :
    input a : UInt<0>
    input b : UInt<8>
    output c : UInt<8>

    c <= mul(a, b)
```

Mid:
```
circuit Foo :
  module Foo :
    input b : UInt<8>
    output c : UInt<8>

    c <= mul(UInt<1>("h0"), b)
```

Low:
```
circuit Foo :
  module Foo :
    input b : UInt<8>
    output c : UInt<8>

    node _GEN_0 = mul(UInt<8>("h0"), b)
    c <= bits(_GEN_0, 7, 0)
```

Verilog:
```verilog
module Foo(
  input  [7:0] b,
  output [7:0] c
);
  wire [8:0] _GEN_0 = 8'h0 * b;
  assign c = _GEN_0[7:0];
endmodule
```

lattner, 2021-01-10:
Awesome, thank you for the help.  Concat and reductions happen to be added 4d89644

lattner, 2021-01-10:
0176805 is enough to support this specific issue, but there is more work on zero width types to go.  Thank you for filing these Aliaksei!

closed 2021-01-10 by lattner

lattner, 2021-01-10:
Ok, I think that all the zero length issues with primitives should be addressed, please let me know.  I will do an audit for wire, node, instance, ... next.