llvm/circt #346: FIRRTL dialect shouldn't depend on RTL

lattner, 2020-12-19:
@mikeurbach points out that this patch:
https://github.com/llvm/circt/commit/73dbad3377c3fc17bdbc94c926c4d7e3023f6ff8#diff-6be8f16b249664b12c96f16587df5fbae2167725d4df3b2d04e075610fcbc9c9R1329

Added a dependency from FIRRTL to RTL because `AnalogInOutCastOp` depends on `InOutType`.  It would be very nice (from a library dependency standpoint) for FIRRTL to not do this.  I'm not sure the best solution for this though, should we pull `AnalogInOutCastOp` out to a different library?  It is an artifact of `LowerToRTL`, so this is probably the right thing to do, but it seems like massive overkill.

mikeurbach, 2020-12-19:
I'm not sure the best way to layer this either but we can't be the first people to cast from a type in one dialect to a type in another. I wonder if others have a way to do this with tying the dialects' libraries together.

youngar, 2022-07-06:
I think we've departed from this design point, FIRRTL uses a couple attributes attributes (e.g. `hw.outputFile`) and operations (e.g. `sv.verbatim`) from other dialects.

closed 2022-07-06 by youngar