llvm/circt #347: [FIRRTL] Negative Signed Literal error

drom, 2020-12-22:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_1: UInt<668>
    output tmp12: SInt<672>
    tmp12 <= mul(SInt<4>(-4), asSInt(inp_1))
```
Compiled with `firtool --lower-to-rtl` produces this error:

```python
top_mod.fir:5:26: error: expected '.' in floating point literal
    tmp12 <= mul(SInt<4>(-4), asSInt(inp_1))
                         ^
```

seldridge, 2020-12-22:
I ran into this a while back with negative external module parameters: https://github.com/llvm/circt/issues/183

We need to both fix the parsing and come up with how to represent the literals.

lattner, 2020-12-25:
Fixed in 362a282 thank you again @drom!

closed 2020-12-25 by lattner