llvm/circt #376: Windows test failures

teqdruid, 2020-12-29:
We have some test failures on Windows. In addition to all the LLHD sim tests (due to `llhd-sim` not being built #374), the following fail:
```
********************
Failed Tests (12):
  CIRCT :: Dialect/FIRRTL/import-locations.fir
  CIRCT :: ExportVerilog/verilog-basic.fir
  CIRCT :: handshake-runner/simple_loop.mlir
```

teqdruid, 2020-12-29:
```
 python.exe ..\llvm\build\Release\bin\llvm-lit.py .\test\Dialect\FIRRTL\import-locations.fir -a
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\config.py:42: note: using lit tools: C:\Program Files\Git\usr\bin
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\subst.py:127: note: Did not find llhd-sim in C:/code/circt/build/bin;;C:/code/circt/llvm/build/Release/bin
-- Testing: 1 tests, 1 workers --
FAIL: CIRCT :: Dialect/FIRRTL/import-locations.fir (1 of 1)
******************** TEST 'CIRCT :: Dialect/FIRRTL/import-locations.fir' FAILED ********************
Script:
--
: 'RUN: at line 1';   c:\code\circt\build\bin\circt-translate.exe -parse-fir --mlir-print-debuginfo -verify-diagnostics -mlir-print-local-scope C:\code\circt\test\Dialect\FIRRTL\import-locations.fir | c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\Dialect\FIRRTL\import-locations.fir
--
Exit Code: 1

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "c:\code\circt\build\bin\circt-translate.exe" "-parse-fir" "--mlir-print-debuginfo" "-verify-diagnostics" "-mlir-print-local-scope" "C:\code\circt\test\Dialect\FIRRTL\import-locations.fir"
$ "c:\code\circt\llvm\build\release\bin\filecheck.exe" "C:\code\circt\test\Dialect\FIRRTL\import-locations.fir"
# command stderr:
C:\code\circt\test\Dialect\FIRRTL\import-locations.fir:27:11: error: CHECK: expected string not found in input
 ; CHECK: } loc({{.*}}test/Dialect/FIRRTL/import-locations.fir"
          ^
<stdin>:9:19: note: scanning from here
 firrtl.module @M3(%in: !firrtl.uint) {
                  ^
<stdin>:10:10: note: possible intended match here
 } loc("C:\code\circt\test\Dialect\FIRRTL\import-locations.fir":25:3)
         ^

Input file: <stdin>
Check file: C:\code\circt\test\Dialect\FIRRTL\import-locations.fir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
            1: module {
            2:  firrtl.circuit "MyModule" {
            3:  firrtl.module @MyModule(%in: !firrtl.uint, %out: !firrtl.flip<uint<8>>) {
            4:  firrtl.connect %out, %in : !firrtl.flip<uint<8>>, !firrtl.uint loc("Somewhere.galaxy":42:1)
            5:  firrtl.skip loc("SKipLoc.scala":42:24)
            6:  } loc("FooBar.scala":369:27)
            7:  firrtl.module @M2(%in: !firrtl.uint) {
            8:  } loc("Filename With Space.scala":1:2)
            9:  firrtl.module @M3(%in: !firrtl.uint) {
check:27'0                       X~~~~~~~~~~~~~~~~~~~~ error: no match found
           10:  } loc("C:\code\circt\test\Dialect\FIRRTL\import-locations.fir":25:3)
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
check:27'1              ?                                                            possible intended match
           11:  firrtl.module @Subexpressions(%in: !firrtl.uint, %auto: !firrtl.flip<bundle<out_0: uint<1>>>) {
check:27'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           12:  %out_0 = firrtl.wire {name = "out_0"} : !firrtl.bundle<member: bundle<0: bundle<reset: uint<1>>>> loc("C:\code\circt\test\Dialect\FIRRTL\import-locations.fir":36:5)
           13:  %0 = firrtl.subfield %auto("out_0") : (!firrtl.flip<bundle<out_0: uint<1>>>) -> !firrtl.flip<uint<1>> loc("Field":173:49)
           14:  %1 = firrtl.subfield %out_0("member") : (!firrtl.bundle<member: bundle<0: bundle<reset: uint<1>>>>) -> !firrtl.bundle<0: bundle<reset: uint<1>>> loc("Field":173:49)
           15:  %2 = firrtl.subfield %1("0") : (!firrtl.bundle<0: bundle<reset: uint<1>>>) -> !firrtl.bundle<reset: uint<1>> loc("Field":173:49)
            .
            .
            .
>>>>>>

error: command failed with exit status: 1

--

********************
********************
Failed Tests (1):
  CIRCT :: Dialect/FIRRTL/import-locations.fir

Testing Time: 0.27s
  Failed: 1
```

teqdruid, 2020-12-29:
```
> python.exe ..\llvm\build\Release\bin\llvm-lit.py .\test\ExportVerilog\verilog-basic.fir -a
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\config.py:42: note: using lit tools: C:\Program Files\Git\usr\bin
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\subst.py:127: note: Did not find llhd-sim in C:/code/circt/build/bin;;C:/code/circt/llvm/build/Release/bin
-- Testing: 1 tests, 1 workers --
FAIL: CIRCT :: ExportVerilog/verilog-basic.fir (1 of 1)
******************** TEST 'CIRCT :: ExportVerilog/verilog-basic.fir' FAILED ********************
Script:
--
: 'RUN: at line 6';   c:\code\circt\build\bin\circt-translate.exe -parse-fir --mlir-print-debuginfo C:\code\circt\test\ExportVerilog\verilog-basic.fir | c:\code\circt\build\bin\circt-translate.exe -emit-verilog -verify-diagnostics | c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\ExportVerilog\verilog-basic.fir --strict-whitespace
: 'RUN: at line 9';   c:\code\circt\build\bin\circt-translate.exe -parse-fir --mlir-print-debuginfo C:\code\circt\test\ExportVerilog\verilog-basic.fir | c:\code\circt\build\bin\circt-opt.exe -pass-pipeline='rtl.module(lower-firrtl-to-rtl)' | c:\code\circt\build\bin\circt-translate.exe -emit-verilog -verify-diagnostics
--
Exit Code: 2

Command Output (stdout):
--
$ ":" "RUN: at line 6"
$ "c:\code\circt\build\bin\circt-translate.exe" "-parse-fir" "--mlir-print-debuginfo" "C:\code\circt\test\ExportVerilog\verilog-basic.fir"
$ "c:\code\circt\build\bin\circt-translate.exe" "-emit-verilog" "-verify-diagnostics"
# command stderr:
<stdin>:346:16: error: unexpected error: unknown escape in string literal
#loc0 = loc("C:\code\circt\test\ExportVerilog\verilog-basic.fir":0:0)
               ^

error: command failed with exit status: 1
$ "c:\code\circt\llvm\build\release\bin\filecheck.exe" "C:\code\circt\test\ExportVerilog\verilog-basic.fir" "--strict-whitespace"
# command stderr:
FileCheck error: '<stdin>' is empty.
FileCheck command line:  c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\ExportVerilog\verilog-basic.fir --strict-whitespace

error: command failed with exit status: 2

--

********************
********************
Failed Tests (1):
  CIRCT :: ExportVerilog/verilog-basic.fir

Testing Time: 0.25s
  Failed: 1
```

teqdruid, 2020-12-29:
> python.exe ..\llvm\build\Release\bin\llvm-lit.py .\test\handshake-runner\simple_loop.mlir
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\config.py:42: note: using lit tools: C:\Program Files\Git\usr\bin
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\subst.py:127: note: Did not find llhd-sim in C:/code/circt/build/bin;;C:/code/circt/llvm/build/Release/bin
-- Testing: 1 tests, 1 workers --
FAIL: CIRCT :: handshake-runner/simple_loop.mlir (1 of 1)
********************
Failed Tests (1):
  CIRCT :: handshake-runner/simple_loop.mlir

Testing Time: 0.36s
  Failed: 1
C:\code\circt\build [windows-build ≡ +1 ~0 -0 !]> python.exe ..\llvm\build\Release\bin\llvm-lit.py .\test\handshake-runner\simple_loop.mlir -a
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\config.py:42: note: using lit tools: C:\Program Files\Git\usr\bin
llvm-lit.py: C:\code\circt\llvm\llvm\utils\lit\lit\llvm\subst.py:127: note: Did not find llhd-sim in C:/code/circt/build/bin;;C:/code/circt/llvm/build/Release/bin
-- Testing: 1 tests, 1 workers --
FAIL: CIRCT :: handshake-runner/simple_loop.mlir (1 of 1)
******************** TEST 'CIRCT :: handshake-runner/simple_loop.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   mlir-opt --convert-std-to-llvm C:\code\circt\test\handshake-runner\simple_loop.mlir | mlir-cpu-runner --entry-point-result=i64 | c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\handshake-runner\simple_loop.mlir
: 'RUN: at line 2';   c:\code\circt\build\bin\circt-opt.exe -create-dataflow C:\code\circt\test\handshake-runner\simple_loop.mlir | c:\code\circt\build\bin\handshake-runner.exe | c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\handshake-runner\simple_loop.mlir
: 'RUN: at line 3';   c:\code\circt\build\bin\handshake-runner.exe C:\code\circt\test\handshake-runner\simple_loop.mlir | c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\handshake-runner\simple_loop.mlir
--
Exit Code: 2

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "mlir-opt" "--convert-std-to-llvm" "C:\code\circt\test\handshake-runner\simple_loop.mlir"
$ "mlir-cpu-runner" "--entry-point-result=i64"
# command stderr:
Assertion failed: (KV.second.getFlags() & ~WeakFlags) == (I->second & ~WeakFlags) && "Resolving symbol with incorrect flags", file C:\code\circt\llvm\llvm\lib\ExecutionEngine\Orc\Core.cpp, line 2633
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: mlir-cpu-runner --entry-point-result=i64
 #0 0x00007ff75c7b0ad5 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x70ad5)
 #1 0x00007ff905841891 (C:\WINDOWS\System32\ucrtbase.dll+0x71891)
 #2 0x00007ff905842861 (C:\WINDOWS\System32\ucrtbase.dll+0x72861)
 #3 0x00007ff9058441c5 (C:\WINDOWS\System32\ucrtbase.dll+0x741c5)
 #4 0x00007ff905844501 (C:\WINDOWS\System32\ucrtbase.dll+0x74501)
 #5 0x00007ff75d298147 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb58147)
 #6 0x00007ff75d2be9df (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb7e9df)
 #7 0x00007ff75d2ba356 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb7a356)
 #8 0x00007ff75d2c99c1 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb899c1)
 #9 0x00007ff75d2bc2d0 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb7c2d0)
#10 0x00007ff75d2ae957 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb6e957)
#11 0x00007ff75d2ac080 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb6c080)
#12 0x00007ff75d2ae321 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb6e321)
#13 0x00007ff75d2ae321 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb6e321)
#14 0x00007ff75d2a9604 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb69604)
#15 0x00007ff75d2a1f7d (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb61f7d)
#16 0x00007ff75d29854d (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb5854d)
#17 0x00007ff75d29bd52 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb5bd52)
#18 0x00007ff75d296f11 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb56f11)
#19 0x00007ff75d299c21 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb59c21)
#20 0x00007ff75d2966ba (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb566ba)
#21 0x00007ff75d2a13b9 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb613b9)
#22 0x00007ff75d2a078c (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb6078c)
#23 0x00007ff75d2a0b26 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb60b26)
#24 0x00007ff75d286724 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0xb46724)
#25 0x00007ff75e37800f (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x1c3800f)
#26 0x00007ff75cb2bb06 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x3ebb06)
#27 0x00007ff75cb281d9 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x3e81d9)
#28 0x00007ff75cb2a70d (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x3ea70d)
#29 0x00007ff75c7a4911 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x64911)
#30 0x00007ff75ddfbbb4 (c:\code\circt\llvm\build\release\bin\mlir-cpu-runner.exe+0x16bbbb4)
#31 0x00007ff906617034 (C:\WINDOWS\System32\KERNEL32.DLL+0x17034)
#32 0x00007ff907b9d0d1 (C:\WINDOWS\SYSTEM32\ntdll.dll+0x4d0d1)

error: command failed with exit status: 2147483651
$ "c:\code\circt\llvm\build\release\bin\filecheck.exe" "C:\code\circt\test\handshake-runner\simple_loop.mlir"
# command stderr:
FileCheck error: '<stdin>' is empty.
FileCheck command line:  c:\code\circt\llvm\build\release\bin\filecheck.exe C:\code\circt\test\handshake-runner\simple_loop.mlir

error: command failed with exit status: 2

--

********************
********************
Failed Tests (1):
  CIRCT :: handshake-runner/simple_loop.mlir

Testing Time: 0.37s
  Failed: 1

lattner, 2020-12-29:
I'd recommend changing the check lines to use . instead of a hardcoded slash, e.g.:
; CHECK: } loc({{.*}}test.Dialect.FIRRTL.import-locations.fir"

teqdruid, 2020-12-29:
Yeah, the first two are relatively simple. Unfortunately, I've only got the stack trace for the release build for the last one.

teqdruid, 2021-07-26:
Update: this list has grown and the firrtl tests fail non-deterministically.

teqdruid, 2021-07-27:
@fabianschuiki This line https://github.com/llvm/circt/blob/main/test/ExportVerilog/output-file.mlir#L41 breaks the test on Windows.

teqdruid, 2021-07-27:
@lattner This line is broken on windows: https://github.com/llvm/circt/blob/d0bfdbc2d6fad89c0f6a9ea0880bc7f3edfba6d0/test/Dialect/FIRRTL/parse-locations.fir#L30
Locations are printed `loc("C:\\code\\circt\\test\\Dialect\\FIRRTL\\parse-locations.fir":28:10)` in Windows. If I move the directory separator from `{{.}}` to `{{..?}}` it works. Is this how you'd like to handle these situations?

teqdruid, 2021-07-28:
@lattner For the `firtool` tests failing non-deterministically: if I add `--mlir-disable-threading` to the invocation in `style.fir`, it no longer crashes. I have seen `firtool/split-verilog.mlir`, `firtool/firtool.mlir`, `firtool/optimizations.fir`, and `firtool/style.fir` fail non-deterministically. In the typical multi threading Hisenbug fashion, I can't make them crash on their own -- only when they run in parallel via lit.

When they do crash, this is the signature:

```
FAIL: CIRCT :: firtool/style.fir (8 of 8)
******************** TEST 'CIRCT :: firtool/style.fir' FAILED ********************
Script:
--
: 'RUN: at line 1';   c:\code\circt\build\bin\firtool.exe C:\code\circt\test\firtool\style.fir | c:\code\circt\llvm\build\debug\bin\filecheck.exe C:\code\circt\test\firtool\style.fir --check-prefix=DEFAULT
: 'RUN: at line 2';   not c:\code\circt\build\bin\firtool.exe --lowering-options=bad-option C:\code\circt\test\firtool\style.fir 2>&1 | c:\code\circt\llvm\build\debug\bin\filecheck.exe C:\code\circt\test\firtool\style.fir --check-prefix=BADOPTION
: 'RUN: at line 3';   c:\code\circt\build\bin\firtool.exe --lowering-options=alwaysFF C:\code\circt\test\firtool\style.fir | c:\code\circt\llvm\build\debug\bin\filecheck.exe C:\code\circt\test\firtool\style.fir --check-prefix=ALWAYSFF
--
Exit Code: 3221225477

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "c:\code\circt\build\bin\firtool.exe" "C:\code\circt\test\firtool\style.fir"
$ "c:\code\circt\llvm\build\debug\bin\filecheck.exe" "C:\code\circt\test\firtool\style.fir" "--check-prefix=DEFAULT"
$ ":" "RUN: at line 2"
$ "not" "c:\code\circt\build\bin\firtool.exe" "--lowering-options=bad-option" "C:\code\circt\test\firtool\style.fir"
$ "c:\code\circt\llvm\build\debug\bin\filecheck.exe" "C:\code\circt\test\firtool\style.fir" "--check-prefix=BADOPTION"
$ ":" "RUN: at line 3"
$ "c:\code\circt\build\bin\firtool.exe" "--lowering-options=alwaysFF" "C:\code\circt\test\firtool\style.fir"
note: command had no output on stdout or stderr
error: command failed with exit status: 3221225477
$ "c:\code\circt\llvm\build\debug\bin\filecheck.exe" "C:\code\circt\test\firtool\style.fir" "--check-prefix=ALWAYSFF"
```

I'm also seeing `firtool/blackbox.mlir` fail, but deterministically with a different signature.

teqdruid, 2021-07-28:
Once the tests which I called out recently are cleared, we will have a Windows build and test which passes!

lattner, 2021-07-28:
Path fix seems totally fine, committed.  I'd really rather not turn off threading for random tests to work around this. Can we xfail: windows these tests until they get root caused?

teqdruid, 2021-07-29:
`xfail: windows` will mark the tests as failing to fail (`Unexpectedly Passed`) in the case where a test passes, which is reasonably often. If I set the `ALLOW_RETRIES: 5` option, they try again and pass. Should I do that?

teqdruid, 2021-07-29:
I spoke too soon. Setting `ALLOW_RETRIES` to an arbitrarily large value doesn't work as consistently as I would like and expected.

teqdruid, 2021-11-02:
Now that we (Microsoft) are actively using CIRCT on Windows, we need to get the Windows tests enabled in the CI flow. (So that we don't break the Windows build silently and don't notice it for a week *cough* #2069 *cough*.)

Here are the tests which were failing as of a week ago (before #2069) in one run on my notebook:

```
Failed Tests (14):
  CIRCT :: Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/async-reset-errors.mlir
  CIRCT :: Dialect/FIRRTL/emit-metadata.mlir
  CIRCT :: Dialect/FIRRTL/expand-whens-errors.mlir
  CIRCT :: Dialect/FIRRTL/expand-whens.mlir
  CIRCT :: Dialect/FIRRTL/grand-central.mlir
  CIRCT :: circt-reduce/trivial.mlir
  CIRCT :: firtool/blackbox.mlir
  CIRCT :: firtool/chirrtl.fir
  CIRCT :: firtool/firtool.fir
  CIRCT :: firtool/memoryLowering.fir
  CIRCT :: firtool/memoryMetadata.fir
  CIRCT :: firtool/optimizations.fir
  CIRCT :: firtool/split-verilog.mlir
```

They're mostly FIRRTL tests, which aren't currently part of my use case.  As such, I'm happy to just `UNSUPPORTED: windows` them. @lattner Once #2069 is fixed, how should we proceed? Only some are due to the threading bug we discussed above (all the `firtool/*` ones). The rest don't seem to be (and may just be directory separator issues). I've attached a log of a run to aid in debugging. [windows_failures.txt](https://github.com/llvm/circt/files/7457940/windows_failures.txt)

seldridge, 2021-11-02:
@teqdruid: Do you have any instructions for how to setup and run CIRCT test inside Visual Studio?

teqdruid, 2021-11-02:
The closest that exists is the windows build GH workflow definition. On
your local machine, it's pretty similar but you've got to use the VS
developer prompt.

 I don't use the VS GUI.

teqdruid, 2021-11-02:
I'll write up a bit on getting my env (msvc + ninja + vscode) set up later
today.

teqdruid, 2021-11-03:
@seldridge I've added some notes on setting up Windows builds: https://github.com/llvm/circt/blob/main/docs/GettingStarted.md#windows-notes-on-setting-up-with-ninja

darthscsi, 2021-11-03:
I'm setting up a windows env.

seldridge, 2021-11-03:
@teqdruid: Thanks! This will likely be helpful for others. I spent 15 minutes trying to figure out how to get a Visual Studio environment on a Mac (or just get access to `msvc`) and gave up. 😂  Lenharth has this sorted out.

youngar, 2022-10-20:
I think that we can close this now after https://github.com/llvm/circt/pull/4110. There are two issues remaining on Windows and I opened individual issues for each: https://github.com/llvm/circt/issues/4128, https://github.com/llvm/circt/issues/4129.

closed 2022-10-20 by youngar

teqdruid, 2022-10-20:
Sounds good! Thanks again all!