llvm/circt #377: Consider generating `iff` in always blocks

lattner, 2020-12-29:
Verilog has syntax to allow writing things like:

`always @(clock iff someValue == 1) `

Which is a combination of an `sv.always` and `sv.if` block, which we frequently generate.  As part of polishing the Verilog output we should decide if it is a good idea to fold these.

teqdruid, 2020-12-29:
As usual with constructs which are less often used, we'd need to ensure
that this syntax works with a variety of simulators and synthesis engines,
including verilator. Should we write up a list of products in which
generated verilog should strive to work?

lattner, 2020-12-29:
Yes, it would be great to build that catalog.  I'd recommend starting with [this list](https://symbiflow.github.io/sv-tests/) which probably isn't enough, but is interesting and useful.

I don't think that we (as a circt community) are likely to find a "one size fits" all answer here.  I think we should have a "default" that is what I think you're arguing for, but then support different verilog dialects (not speaking about the MLIR sense of 'dialect' here) for tools that are only targeting a subset of tools.  Those verilog dialects can even include tool specific verilog extensions etc.

An analogy is the X86 backend which supports different subtargets, e.g. SSE1 vs SSE2 vs AMD extensions etc.

teqdruid, 2020-12-29:
Agreed on all fronts.