llvm/circt #394: [FIRRTL] (add / mux) missmatch

drom, 2021-01-04:
## 1) The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_3: SInt<16>
    input inp_6: UInt<13>
    output tmp13: SInt<14>
    wire tmp11: UInt<14>
    tmp11 <= add(inp_6, UInt<12>(1))
    tmp13 <= mux(tail(asUInt(inp_3), 15), asSInt(tmp11), SInt<13>(1))
```

Compiled with `firtool --lower-to-rtl` produces this Verilog:

```verilog
module top_mod(
  input  [15:0] inp_3,
  input  [12:0] inp_6,
  output [13:0] tmp13);

  wire [13:0] tmp11;	// a_top_mod.fir:6:5

  assign tmp11 = {{1'd0}, inp_6} + 14'h1;	// a_top_mod.fir:7:{11,14}
  assign tmp13 = inp_3[0] ? tmp11 : 14'h1;	// a_top_mod.fir:2:3, :8:{14,18,43}
endmodule
```

Compiled with `firrtl-1.4.0` produces this Verilog:

```verilog
module top_mod(
  input  [15:0] inp_3,
  input  [12:0] inp_6,
  output [13:0] tmp13
);
  wire [15:0] _GEN_0 = inp_3;
  assign tmp13 = _GEN_0[0] ? $signed(inp_6 + 13'h1) : $signed(14'sh1);
endmodule
```

Yosys (Yosys 0.9+3755 (git sha1 442d19f6, clang 11.0.0 -fPIC -Os)) reports formal mismatch:

```
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

lattner, 2021-01-08:
Interesting, thanks for the bug report.  I infer that there is something different between `$signed(inp_6 + 13'h1)` and `{{1'd0}, inp_6} + 14'h1` but I don't see what it is.  Do you know what input they differ on?  I think I'm not understanding the order of operations of `$signed(inp_6 + 13'h1)`.

lattner, 2021-01-10:
Any thoughts here @drom, @darthscsi , @seldridge ?

darthscsi, 2021-01-11:
inp_6 + 13'h1 is a 13 bit addition which might overflow, {{1'd0}, inp_6} + 14'h1 is a 14 bit addition which won't.

lattner, 2021-01-11:
Ok, please be patient with me.  Is this reduced testcase the same?

```
circuit top_mod :
  module top_mod :
    input inp_6: UInt<13>
    output tmp13: SInt<14>
    tmp13 <= asSInt(add(inp_6, UInt<12>(1)))
```

Or is there something weird going on with the ternary?  This currently lowers into this FIRRTL:

```
module  {
  firrtl.circuit "top_mod" {
    firrtl.module @top_mod(%inp_6: !firrtl.uint<13>, %tmp13: !firrtl.flip<sint<14>>) {
      %c1_ui12 = firrtl.constant(1 : ui12) : !firrtl.uint<12>
      %0 = firrtl.add %inp_6, %c1_ui12 : (!firrtl.uint<13>, !firrtl.uint<12>) -> !firrtl.uint<14>
      %1 = firrtl.asSInt %0 : (!firrtl.uint<14>) -> !firrtl.sint<14>
      firrtl.connect %tmp13, %1 : !firrtl.flip<sint<14>>, !firrtl.sint<14>
    }
  }
}
```

then to this RTL:

```
 rtl.module @top_mod(%inp_6: i13) -> (%tmp13: i14) {
    %false = rtl.constant(false) : i1
    %c1_i14 = rtl.constant(1 : i14) : i14
    %0 = rtl.concat %false, %inp_6 : (i1, i13) -> i14
    %1 = rtl.add %0, %c1_i14 : i14
    rtl.output %1 : i14
  }
```

Then to this verilog:

```
module top_mod(
  input  [12:0] inp_6,
  output [13:0] tmp13);

  assign tmp13 = {1'h0, inp_6} + 14'h1;	// y.fir:2:3, :5:21
endmodule
```

darthscsi, 2021-01-11:
yea, sorry for the terseness, power was out.  The lower-to-rtl is doing a zero extend, then a wider add, while the firrtl is doing a narrower add then a sign extend.
This might be an issue in the firrtl sign-extend logic.  It really looks like it should do the unsigned add at 14 bit then the asSInt is a noop (bitwise).  But it is definately at 13bit add and then a sign extend.
FIRRTL had a bug in it's sign-extend logic for neg which looked a lot like this.  Namely, it was producing ```x = $signed(y op z)```  expecting width(op) == width(x), which it didn't (op is narrower, result is extended to match x).  If you have ```x = y op z```, then width(op) == width(x) (y and z are extended to match x).

lattner, 2021-01-11:
Yeah, I am pretty sure that this is a SFC bug.  @seldridge what do you think?

lattner, 2021-01-11:
I note that Aliaksei is using firrtl-1.4.0, I wonder if this is different on top of tree

seldridge, 2021-01-11:
No change in 1.4.1.

I'd like to know what the bit width of `$signed(inp_6 + 13'h1)` actually is. If this is 13-bit, and not coerced to 14-bit because of the LHS width, then I think this is an SFC bug. If this is 14-bit, then it's fine. (However, I have a very hard time eyeballing this without having the context-dependent rules next to me...)

Also note that I don't think `$signed` does a sign extension, just interprets as signed. Therefore, due to Verilog ternary x-propagation (and assuming that add is 13-bit), formal is probably catching that `tmp13` is, for an `x` condition, `{1'b0, 13'bx}` in the SFC and `14'bx` in the MFC. It would help if we could spin up a simulator to show this, though.

seldridge, 2021-01-11:
Note: this is actually very weird now that I think about it...

I think there's a bug, but this was not something that was ever considered or should happen when coming from Chisel. Chisel and the SFC will start with all expressions broken out as nodes. The SFC understands self-determined and context-determined primops and isn't supposed to inline if it will result in a Verilog bug like this. However, the input FIRRTL _is already inlined and the SFC isn't setup to chunk expressions back into nodes if there's a problem._ So, yes, this is an SFC bug, but it may not be worthwhile to fix.

Super interesting find @drom.

To clarify, if the following was generated from Chisel, it would follow the path below. The following Chisel circuit is approximately the same thing as the bad code:

```scala
class top_mod extends RawModule {
  val a = IO(Input(Bool()))
  val inp_6 = IO(Input(UInt(13.W)))
  val tmp13 = IO(Output(SInt(14.W)))
  
  val tmp11 = Wire(UInt(14.W))
  tmp11 := inp_6 +& 1.U(12.W) // +& is width expanding, like FIRRTL
  tmp13 := Mux(a, tmp11.asSInt, 1.S(13.W))
}
```

This is emitted out of Chisel with all nodes split out:

```
circuit top_mod :
  module top_mod :
    input a : UInt<1>
    input inp_6 : UInt<13>
    output tmp13 : SInt<14>
    wire tmp11 : UInt<14>
    node _tmp11_T = add(inp_6, UInt<12>("h1"))
    tmp11 <= _tmp11_T
    node _tmp13_T = asSInt(tmp11)
    node _tmp13_T_1 = mux(a, _tmp13_T, asSInt(UInt<13>("h1")))
    tmp13 <= _tmp13_T_1
```

This will then produce the following Verilog:

```
module top_mod(
  input         a,
  input  [12:0] inp_6,
  output [13:0] tmp13
);
  wire [13:0] _tmp13_T = inp_6 + 13'h1; 
  assign tmp13 = a ? $signed(_tmp13_T) : $signed(14'sh1); 
endmodule
```

This all looks good. However, if I change the input FIRRTL IR to inline the `asSint`... everything falls apart:

```
circuit Foo :
  module Foo :
    input a : UInt<1>
    input inp_6 : UInt<13>
    output tmp13 : SInt<14>
    wire tmp11 : UInt<14>
    node _tmp11_T = add(inp_6, UInt<12>("h1"))
    tmp11 <= _tmp11_T
    node _tmp13_T_1 = mux(a, asSInt(tmp11), asSInt(UInt<13>("h1")))
    tmp13 <= _tmp13_T_1
```

This produces:

```verilog
module Foo(
  input         a,
  input  [12:0] inp_6,
  output [13:0] tmp13
);
  assign tmp13 = a ? $signed(inp_6 + 13'h1) : $signed(14'sh1);
endmodule
```

I'll file an SFC issue, but this may not get fixed.

@drom: It might be good to add an option to your fuzzer to start with all expressions broken out into nodes. That would work around this issue...

lattner, 2021-01-11:
Yeah, I think it is clear that this is [a SFC problem](https://github.com/chipsalliance/firrtl/issues/2037) at this point.  The inlining is killing an extension that otherwise happen due to the assignment to the wider wire.  I'm going to close this as "behaves correctly".  We don't want to be "bug for bug" compatible here, because there is no reasonable way to emulate this.

closed 2021-01-11 by lattner