llvm/circt #395: firrtl.node type verification

darthscsi, 2021-01-04:
The following now fails

```
module  {
  firrtl.circuit "foo" {
    firrtl.extmodule @bar(%baz: !firrtl.uint<1>)
    firrtl.module @foo() {
      %fetch = firrtl.instance @bar {name = "fetch"} : !firrtl.bundle<baz: flip<uint<1>>>
      %1454 = firrtl.subfield %fetch("baz") : (!firrtl.bundle<baz: flip<uint<1>>>) -> !firrtl.flip<uint<1>>
      %hits_1_7 = firrtl.node %1454 {name = "hits_1_7"} : !firrtl.flip<uint<1>>
    }
  }
}
```

seldridge, 2021-01-05:
This may be fixed with #381. Relevant code: https://github.com/llvm/circt/pull/381/files#diff-ee090cee6cf72cf93f7f3c6685d8a29e8d5b79fee0e3d68691ba10f32c1da42aR2118

seldridge, 2021-01-07:
I think there's a couple of things going on here:

1. The FIRRTL spec requires that nodes are passive. We should add a verifier to enforce this so that this fails earlier than during conversion to the RTL dialect.
2. CHIRRTL (and the SFC) accept flipped nodes that are passive under the flip, e.g., you can have a node pointing at an output port of the current module.

(1) should be disallowed via verification of the node op. I'll look into adding a verifier for this.

(2) should be fixed up via #381 where a node passive under a single outer flip is converted to passive via an `asPassive` operand. #381 implements this.

PR #416 (2021-01-08): [FIRRTL] Verify that NodeOp is passive
Add a NodeOp verifier that asserts that a node is passive. One
testcase is added and one testcase is changed that did not respect
this invariant.

- Fixes #395.
- Fixes #388.
closed 2021-01-08 by seldridge