llvm/circt #397: [ExportVerilog] Emit integer signedness in type

teqdruid, 2021-01-05:
Emit the signedness whenever the type is printed, especially in structs.

```verilog
struct example {
  logic signed [7:0] a; // Type si8
  logic unsigned [7:0] a; // Type ui8
  logic [7:0] a; // Type i8
}
```

This should probably be based on the IntegerType's signedness so it works correctly in the case where there are no operations on said signal/port to be found.

teqdruid, 2021-01-05:
ESI will be producing structs for people hand writing SV. They don't want to be wrapping their operators in `$signed(ex.a)`.

lattner, 2021-01-05:
If this is important to preserve in the type system then we'll probably need an `sv.struct` type.

lattner, 2021-01-05:
Or some way to bound how signed types where and where they occur (e.g. have a cast from signfull to signless and back).